AxisCommunications/acap-native-sdk-examples #13: no overlay on the camera stream

pppaulpeter, 2021-11-23:
[image]
i have install the example on the testing camera and running, but nothing showing on the live stream. it should display some bounding boxes and text overlay,right? 
this is the camera log:
----- Contents of SYSTEM_LOG for 'axoverlay' -----

2021-11-23T12:49:08.163+00:00 axis-b8a44f0e633b [ WARNING ] axoverlay[16324]: g_error_free: assertion 'error != NULL' failed
2021-11-23T12:49:08.319+00:00 axis-b8a44f0e633b [ WARNING ] axoverlay[1005]: Last message 'g_error_free: assert' repeated 1 times, suppressed by syslog-ng on axis-b8a44f0e633b
2021-11-23T12:50:09.065+00:00 axis-b8a44f0e633b [ WARNING ] axoverlay[16857]: g_error_free: assertion 'error != NULL' failed
2021-11-23T12:50:09.319+00:00 axis-b8a44f0e633b [ WARNING ] axoverlay[1005]: Last message 'g_error_free: assert' repeated 1 times, suppressed by syslog-ng on axis-b8a44f0e633b
2021-11-23T12:53:07.446+00:00 axis-b8a44f0e633b [ WARNING ] axoverlay[17954]: g_error_free: assertion 'error != NULL' failed
2021-11-23T12:53:08.319+00:00 axis-b8a44f0e633b [ WARNING ] axoverlay[1005]: Last message 'g_error_free: assert' repeated 1 times, suppressed by syslog-ng on axis-b8a44f0e633b
2021-11-23T13:02:31.447+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Max resolution (width x height): 1920 x 1080
2021-11-23T13:02:31.447+00:00 axis-b8a44f0e633b [ WARNING ] axoverlay[20187]: g_error_free: assertion 'error != NULL' failed
2021-11-23T13:02:32.319+00:00 axis-b8a44f0e633b [ WARNING ] axoverlay[1005]: Last message 'g_error_free: assert' repeated 1 times, suppressed by syslog-ng on axis-b8a44f0e633b
2021-11-23T13:02:37.308+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for overlay: 1920 x 1080
2021-11-23T13:02:37.308+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for stream: 1280 x 720
2021-11-23T13:02:37.417+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for overlay: 1920 x 1080
2021-11-23T13:02:37.417+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for stream: 1280 x 720
2021-11-23T13:02:37.417+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for camera: 1
2021-11-23T13:02:37.417+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for overlay: 1280 x 720
2021-11-23T13:02:37.417+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for stream: 1280 x 720
2021-11-23T13:02:37.574+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for camera: 1
2021-11-23T13:02:37.574+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for overlay: 1280 x 720
2021-11-23T13:02:37.574+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for stream: 1280 x 720
2021-11-23T13:02:37.638+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for overlay: 1920 x 1080
2021-11-23T13:02:37.638+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for stream: 1280 x 720
2021-11-23T13:02:37.638+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for camera: 1
2021-11-23T13:02:37.638+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for overlay: 1280 x 720
2021-11-23T13:02:37.638+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for stream: 1280 x 720
2021-11-23T13:02:37.646+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for overlay: 1920 x 1080
2021-11-23T13:02:37.646+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for stream: 1280 x 720
2021-11-23T13:02:37.653+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for camera: 1
2021-11-23T13:02:37.653+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for overlay: 1280 x 720
2021-11-23T13:02:37.653+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for stream: 1280 x 720
2021-11-23T13:02:39.326+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for overlay: 1920 x 1080
2021-11-23T13:02:39.326+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for stream: 1280 x 720
2021-11-23T13:02:39.326+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for camera: 1
2021-11-23T13:02:39.326+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for overlay: 1280 x 720
2021-11-23T13:02:39.326+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for stream: 1280 x 720
2021-11-23T13:02:39.329+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for overlay: 1920 x 1080
2021-11-23T13:02:39.329+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for stream: 1280 x 720
2021-11-23T13:02:39.334+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for camera: 1
2021-11-23T13:02:39.334+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for overlay: 1280 x 720
2021-11-23T13:02:39.334+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for stream: 1280 x 720
2021-11-23T13:02:40.323+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for overlay: 1920 x 1080
2021-11-23T13:02:40.323+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for stream: 1280 x 720
2021-11-23T13:02:40.325+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for camera: 1
2021-11-23T13:02:40.325+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for overlay: 1280 x 720
2021-11-23T13:02:40.325+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for stream: 1280 x 720
2021-11-23T13:02:40.328+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for overlay: 1920 x 1080
2021-11-23T13:02:40.328+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for stream: 1280 x 720
2021-11-23T13:02:40.336+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for camera: 1
2021-11-23T13:02:40.336+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for overlay: 1280 x 720
2021-11-23T13:02:40.336+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for stream: 1280 x 720
2021-11-23T13:02:41.323+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for overlay: 1920 x 1080
2021-11-23T13:02:41.323+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for stream: 1280 x 720
2021-11-23T13:02:41.323+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for camera: 1
2021-11-23T13:02:41.323+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for overlay: 1280 x 720
2021-11-23T13:02:41.323+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for stream: 1280 x 720
2021-11-23T13:02:41.326+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for overlay: 1920 x 1080
2021-11-23T13:02:41.326+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for stream: 1280 x 720
2021-11-23T13:02:41.334+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for camera: 1
2021-11-23T13:02:41.337+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for overlay: 1280 x 720
2021-11-23T13:02:41.337+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for stream: 1280 x 720
2021-11-23T13:02:42.326+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for overlay: 1920 x 1080
2021-11-23T13:02:42.326+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for overlay: 1920 x 1080
2021-11-23T13:02:42.326+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for stream: 1280 x 720
2021-11-23T13:02:42.326+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for stream: 1280 x 720
2021-11-23T13:02:42.326+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for camera: 1
2021-11-23T13:02:42.326+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for camera: 1
[... 188 lines cut ...]
2021-11-23T13:03:01.324+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for stream: 1280 x 720
2021-11-23T13:03:01.328+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for overlay: 1920 x 1080
2021-11-23T13:03:01.328+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for stream: 1280 x 720
[... cut ...]
2021-11-23T13:03:03.326+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for stream: 1280 x 720
2021-11-23T13:03:03.329+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for overlay: 1920 x 1080
2021-11-23T13:03:03.329+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Adjust callback for stream: 1280 x 720
2021-11-23T13:03:03.340+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for camera: 1
2021-11-23T13:03:03.340+00:00 axis-b8a44f0e633b [ INFO    ] axoverlay[20187]: Render callback for overlay: 1280 x 720

mikaelli-axis, 2021-11-26:
Thank you for your report. What FW version are you using in this case?

pppaulpeter, 2021-12-14:
Hi, 
the firmware version is 
10.7.3

sheh, 2021-12-23:
The same issue on P3255-LVE:
```
root.Properties.Firmware.Version=10.7.3
root.Properties.FirmwareManagement.Version=1.4
```
Is there any solution? Or workaround?

sheh, 2021-12-28:
@pppaulpeter I have solution)
I replaced a PoE injector with a proper PoE switch class 4. If you see red warning about PoE supply on top of the Live View you should do the same. This warning has higher priority than user's overlays. The example works for me now.

mikaelli-axis, 2022-01-03:
@sheh Good catch, I guess we should make sure to ask about PoE in issue templates from now on.

closed 2022-01-26 by pataxis

xsacha, 2022-01-31:
This was a duplicate of https://github.com/AxisCommunications/acap3-examples/issues/114 but didn't catch it because reported in a different repo. Sorry.