AxisCommunications/acap-native-sdk-examples #35: Issue with Larod API

ynkze, 2022-02-23:
Axis product/device: P1375

Device firmware version: AXIS OS version 10.5.2

Issue description: I tried a few examples such as the hello-world and web-server and it works as intended, however when I tried the vdo-larod example, I encounter an issue and this is the log:
/usr/lib/liblarod.so.1: version `LIBLAROD_3.0.0' not found (required by /usr/local/packages/vdo_larod/vdo_larod)
I followed the instruction correctly, not sure how I could fix it or upgrade liblarod.

pataxis, 2022-02-23:
The examples use the latest release ACAP SDK 4.1 where the image `acap-native-sdk` has tag 1.1 and this is compatible with AXIS OS 10.9 or higher. Installing the `vdo-larod` example on AXIS OS 10.9 or higher should solve the problem with a newer version of liblarod not available.

For more information on SDK and AXIS OS compatibility, see the ACAP 4 [compatibilty](https://axiscommunications.github.io/acap-documentation/docs/axis-devices.html) documentation and the [4.1 release notes](https://axiscommunications.github.io/acap-documentation/docs/release-notes/4.1.html).

ynkze, 2022-02-25:
> 

Upgraded my OS version and it was fixed, thanks!

closed 2022-02-25 by ynkze