AxisCommunications/acap-native-sdk-examples #37: Container example not launching

JoniSuominen, 2022-04-07:
### Describe the bug

The container example creates the .EAP file, but it does not launch on the camera.
Logs from system report:
`2022-04-07T14:59:00.775+03:00 axis-b8a44f412671 [ ERR     ] systemd[12116]: sdkcontainerExample.service: Failed at step EXEC spawning /usr/local/packages/containerExample/containerExample: No such file or directory
2022-04-07T14:59:00.775+03:00 axis-b8a44f412671 [ ERR     ] systemd[12116]: sdkcontainerExample.service: Failed to execute /usr/local/packages/containerExample/containerExample: No such file or directory
2022-04-07T15:22:12.977+03:00 axis-b8a44f412671 [ ERR     ] systemd[15164]: sdkcontainerExample.service: Failed at step EXEC spawning /usr/local/packages/containerExample/containerExample: No such file or directory
2022-04-07T15:22:12.977+03:00 axis-b8a44f412671 [ ERR     ] systemd[15164]: sdkcontainerExample.service: Failed to execute /usr/local/packages/containerExample/containerExample: No such file or directory`

I tried running journalctl too, but it didn't give any more information.

### Environment

- Axis device model:  Q1615 Mk III
- Axis device firmware version: 10.10.69

pataxis, 2022-04-12:
Hi @JoniSuominen, thanks for reaching out, we will investigate the issue and get back to you.

marbali8, 2022-04-13:
Hi @JoniSuominen, I can't seem to reproduce your problem. Could you give more information on how to reproduce it? The steps I'm following are:

1. Installing [Docker Compose ACAP](https://github.com/AxisCommunications/docker-compose-acap) on the camera.
2. Building the container example.
3. Adding it to the camera and starting it.
4. Visiting http://AXIS_DEVICE_ID:8080.

With that, I get the following reports:

2022-04-13T08:07:09.756+00:00 axis-b8a44f279ba8 [ INFO    ] acapctl: Successfully uninstalled application 'containerExample'
2022-04-13T08:09:59.642+00:00 axis-b8a44f279ba8 [ INFO    ] acapctl: Successfully installed application 'containerExample'
2022-04-13T08:09:59.665+00:00 axis-b8a44f279ba8 [ INFO    ] packagemanager.cgi: pm_install_acap.c:132 pm_install_acap: Total time to install acap 'containerExample': 3 sec.
2022-04-13T08:10:11.087+00:00 axis-b8a44f279ba8 [ INFO    ] acapctl: Started application containerExample
2022-04-13T08:10:11.429+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: Container containerexample_alpine_1  Recreate
2022-04-13T08:10:11.688+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: Container containerexample_alpine_1  Recreated
2022-04-13T08:10:11.710+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: Attaching to alpine_1
2022-04-13T08:10:11.887+00:00 axis-b8a44f279ba8 [ INFO    ] containerd[887]: time="2022-04-13T08:10:11.885485459Z" level=info msg="starting signal loop" namespace=moby path=/run/containerd/io.containerd.runtime.v2.task/moby/9be9d833e9e855beeb2af14bee75425512e0b0074c645f5eae9ffed96a9e67d9 pid=4608
2022-04-13T08:13:07.906+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: alpine_1  | GET / HTTP/1.1
2022-04-13T08:13:07.906+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: alpine_1  | Host: 192.168.1.185:8080
2022-04-13T08:13:07.906+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: alpine_1  | Upgrade-Insecure-Requests: 1
2022-04-13T08:13:07.906+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: alpine_1  | Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
2022-04-13T08:13:07.906+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: alpine_1  | User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/15.4 Safari/605.1.15
2022-04-13T08:13:07.906+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: alpine_1  | Accept-Language: en-GB,en;q=0.9
2022-04-13T08:13:07.906+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: alpine_1  | Accept-Encoding: gzip, deflate
2022-04-13T08:13:07.906+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: alpine_1  | Connection: keep-alive
2022-04-13T08:13:07.906+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: alpine_1  |
2022-04-13T08:13:08.139+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: alpine_1  | GET /favicon.ico HTTP/1.1
2022-04-13T08:13:08.139+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: alpine_1  | Host: 192.168.1.185:8080
2022-04-13T08:13:08.139+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: alpine_1  | Connection: keep-alive
2022-04-13T08:13:08.139+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: alpine_1  | Accept: */*
2022-04-13T08:13:08.139+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: alpine_1  | User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/15.4 Safari/605.1.15
2022-04-13T08:13:08.139+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: alpine_1  | Accept-Language: en-GB,en;q=0.9
2022-04-13T08:13:08.139+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: alpine_1  | Referer: http://192.168.1.185:8080/
2022-04-13T08:13:08.139+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: alpine_1  | Accept-Encoding: gzip, deflate
2022-04-13T08:13:08.139+00:00 axis-b8a44f279ba8 [ INFO    ] containerExample[4544]: alpine_1  |

JoniSuominen, 2022-04-22:
Hi, sorry for the late response

1. docker pull arm32v7/alpine:3.14.0
2. docker save -o alpine.tar arm32v7/alpine:3.14.0
3. docker build --tag containerexample .
4. docker cp $(docker create containerexample):/opt/app/ ./build
5.  upload to camera and boot up

Systemlogs attached.
[systemlog.txt](https://github.com/AxisCommunications/acap-native-sdk-examples/files/8537829/systemlog.txt)
[serverreport.txt](https://github.com/AxisCommunications/acap-native-sdk-examples/files/8537844/serverreport.txt)

marbali8, 2022-04-22:
Hello,
Is your docker-compose.yml file using "${ALPINE_IMAGE}"? If that's so, could you try to change it to arm32v7/alpine:3.14.0?

JoniSuominen, 2022-04-22:
Same issue still persists when doing that

marbali8, 2022-04-25:
Hello,
I can't manage to reproduce your error, same firmware and same camera. I have looked at the systemlog you sent and compared it with mine and the only relevant different message we get is the one you originally sent. The "EXEC spawning" error can be related to directory permissions, so maybe there is some problem reading alpine.tar. 

Apart from that, other differences between your environment and mine could be:
- Version of code: Make sure that you are also running the last version of the example. 
- What is running on the camera: I am only running this example on the camera.
- Configuration: I tried factory default parameters.

JoniSuominen, 2022-04-27:
Thanks!
I got it working  Issue was likely something to do with Windows formatting end of lines, since I could get the example working on WSL just fine. Maybe related to this https://stackoverflow.com/questions/29140377/sh-file-not-found or something. Can be closed if you dont want to investigate further.

marbali8, 2022-04-28:
Great!

I'm glad you got it working. Thank you for your contribution.

closed 2022-04-28 by pataxis