AxisCommunications/acap-native-sdk-examples #41: expanding the using-opencv example to include blob detection

stephanschulz, 2022-04-22:
I am using the M3115-LVE which can not run docker python code but it can run c++ apps.
I successfully compiled and installed the [using-opencv app](https://github.com/AxisCommunications/acap-native-sdk-examples/tree/master/using-opencv) from your example repo.

I am now hoping to expand on the example by including blob detection via the [OpenCV SimpleBlobDetector](https://stackoverflow.com/questions/8076889/how-to-use-opencv-simpleblobdetector). But it seems some of the needed headers are not found and are probably not included when opencv gets build during `docker build --tag <APP_IMAGE> .`

Here my terminal print out that states `SimpleBlobDetector`  is not a class, namespace.

```
Last login: Wed Apr 20 09:22:51 on ttys001
You have mail.
stephanschulz@Komputer ~ % cd /Users/stephanschulz/Documents/GitHub/axis-camera-computervision/acap-native-sdk-opencv-blob/using-opencv
stephanschulz@Komputer using-opencv % docker build --tag opencv_detector .
[+] Building 2.3s (16/16) FINISHED                                                                                                                 
 => [internal] load build definition from Dockerfile                                                                                          0.0s
 => => transferring dockerfile: 120B                                                                                                          0.0s
 => [internal] load .dockerignore                                                                                                             0.0s
 => => transferring context: 2B                                                                                                               0.0s
 => [internal] load metadata for docker.io/axisecp/acap-native-sdk:1.1-armv7hf-ubuntu20.04                                                    0.3s
 => [ 1/12] FROM docker.io/axisecp/acap-native-sdk:1.1-armv7hf-ubuntu20.04@sha256:0edbfa86b7c8ee82a877d57716898756f3e525f907ee6d964ca3ca45ce  0.0s
 => [internal] load build context                                                                                                             0.0s
 => => transferring context: 5.89kB                                                                                                           0.0s
 => CACHED [ 2/12] COPY sources.list /etc/apt                                                                                                 0.0s
 => CACHED [ 3/12] RUN if [ "armv7hf" = armv7hf ]; then         apt-get update &&         dpkg --add-architecture armhf &&         apt-get u  0.0s
 => CACHED [ 4/12] RUN apt-get update && apt-get install -y -f --no-install-recommends     cmake     curl     pkg-config &&     update-ca-ce  0.0s
 => CACHED [ 5/12] RUN if [ "armv7hf" = armv7hf ]; then         export ARCHDIR=arm-linux-gnueabihf &&         export STRIP=arm-linux-gnueabi  0.0s
 => CACHED [ 6/12] WORKDIR /workspace                                                                                                         0.0s
 => CACHED [ 7/12] RUN curl -fsSL https://github.com/opencv/opencv/archive/4.5.1.tar.gz | tar -xz                                             0.0s
 => CACHED [ 8/12] WORKDIR /workspace/opencv-4.5.1/build                                                                                      0.0s
 => CACHED [ 9/12] RUN COMMON_CMAKE_FLAGS="-D CMAKE_BUILD_TYPE=RELEASE         -D CMAKE_INSTALL_PREFIX=/target-root/usr         -D WITH_OPEN  0.0s
 => [10/12] COPY app/ /opt/app                                                                                                                0.0s
 => [11/12] WORKDIR /opt/app                                                                                                                  0.0s
 => ERROR [12/12] RUN . /opt/axis/acapsdk/environment-setup* && acap-build .                                                                  1.8s
------                                                                                                                                             
 > [12/12] RUN . /opt/axis/acapsdk/environment-setup* && acap-build .:                                                                             
#16 0.320 make                                                                                                                                     
#16 0.322 mkdir -p ./lib                                                                                                                           
#16 0.326 cp /target-root/usr/lib/libopencv_core.so.4.5 /target-root/usr/lib/libopencv_core.so.4.5.1 /target-root/usr/lib/libopencv_core.so ./lib  
#16 0.334 cp /target-root/usr/lib/libopencv_imgproc.so.4.5.1 /target-root/usr/lib/libopencv_imgproc.so /target-root/usr/lib/libopencv_imgproc.so.4.5 ./lib
#16 0.342 cp /target-root/usr/lib/libopencv_video.so /target-root/usr/lib/libopencv_video.so.4.5 /target-root/usr/lib/libopencv_video.so.4.5.1 ./lib
#16 0.349 arm-linux-gnueabihf-g++  -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/opt/axis/acapsdk/sysroots/armv7hf -L/opt/axis/acapsdk/sysroots/armv7hf/usr/lib example.cpp  -O2 -pipe -g -feliminate-unused-debug-types  -Os -pipe -std=c++11 -I/opt/axis/acapsdk/sysroots/armv7hf/usr/include/vdo -I/opt/axis/acapsdk/sysroots/armv7hf/usr/include/gio-unix-2.0 -I/opt/axis/acapsdk/sysroots/armv7hf/usr/include/glib-2.0 -I/opt/axis/acapsdk/sysroots/armv7hf/usr/lib/glib-2.0/include -I/target-root/usr/include/opencv4 -L./lib -Wl,--no-as-needed,-rpath,'$ORIGIN/lib' -lm -lopencv_video -lopencv_imgproc -lopencv_core -lpthread -lvdostream -lsystemd -lgio-2.0 -lgobject-2.0 -lglib-2.0 imgprovider.cpp -o opencv_app && arm-linux-gnueabihf-strip --strip-unneeded opencv_app
#16 1.401 example.cpp: In function ‘int main(int, char**)’:
#16 1.401 example.cpp:101:9: error: ‘SimpleBlobDetector’ has not been declared
#16 1.401   101 |         SimpleBlobDetector::Params params;
#16 1.401       |         ^~~~~~~~~~~~~~~~~~
#16 1.404 example.cpp:104:9: error: ‘params’ was not declared in this scope
#16 1.404   104 |         params.minThreshold = 10;
#16 1.404       |         ^~~~~~
#16 1.416 example.cpp:136:9: error: ‘SimpleBlobDetector’ was not declared in this scope
#16 1.416   136 |     Ptr<SimpleBlobDetector> detector = SimpleBlobDetector::create(params);
#16 1.416       |         ^~~~~~~~~~~~~~~~~~
#16 1.416 example.cpp:136:27: error: template argument 1 is invalid
#16 1.416   136 |     Ptr<SimpleBlobDetector> detector = SimpleBlobDetector::create(params);
#16 1.416       |                           ^
#16 1.416 example.cpp:136:40: error: ‘SimpleBlobDetector’ is not a class, namespace, or enumeration
#16 1.416   136 |     Ptr<SimpleBlobDetector> detector = SimpleBlobDetector::create(params);
#16 1.416       |                                        ^~~~~~~~~~~~~~~~~~
#16 1.416 example.cpp:139:13: error: base operand of ‘->’ is not a pointer
#16 1.416   139 |     detector->detect(fg, keypoints);
#16 1.416       |             ^~
#16 1.417 example.cpp:145:26: warning: too many arguments for format [-Wformat-extra-args]
#16 1.417   145 |         syslog(LOG_INFO, "keypoints", keypoints[0]);
#16 1.417       |                          ^~~~~~~~~~~
#16 1.796 make: *** [Makefile:30: opencv_app] Error 1
#16 1.797 
#16 1.797 make failed. Please fix above errors, before you can create a package
------
executor failed running [/bin/bash -o pipefail -c . /opt/axis/acapsdk/environment-setup* && acap-build .]: exit code: 1
stephanschulz@Komputer using-opencv % 
```

stephanschulz, 2022-04-22:
i tried the following:

add #include <opencv2/features2d.hpp>

edit the Dockerfile to include -D BUILD_LIST=core,imgproc,video,features2d,objdetect \ in lline 76 https://github.com/AxisCommunications/acap-native-sdk-examples/blob/master/using-opencv/Dockerfile#L76

But still get errors
```
 > [12/12] RUN . /opt/axis/acapsdk/environment-setup* && acap-build .:
#16 0.333 make
#16 0.335 mkdir -p ./lib
#16 0.342 cp /target-root/usr/lib/libopencv_core.so.4.5 /target-root/usr/lib/libopencv_core.so.4.5.1 /target-root/usr/lib/libopencv_core.so ./lib
#16 0.350 cp /target-root/usr/lib/libopencv_imgproc.so.4.5.1 /target-root/usr/lib/libopencv_imgproc.so /target-root/usr/lib/libopencv_imgproc.so.4.5 ./lib
#16 0.359 cp /target-root/usr/lib/libopencv_video.so /target-root/usr/lib/libopencv_video.so.4.5 /target-root/usr/lib/libopencv_video.so.4.5.1 ./lib
#16 0.367 arm-linux-gnueabihf-g++  -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/opt/axis/acapsdk/sysroots/armv7hf -L/opt/axis/acapsdk/sysroots/armv7hf/usr/lib example.cpp  -O2 -pipe -g -feliminate-unused-debug-types  -Os -pipe -std=c++11 -I/opt/axis/acapsdk/sysroots/armv7hf/usr/include/vdo -I/opt/axis/acapsdk/sysroots/armv7hf/usr/include/gio-unix-2.0 -I/opt/axis/acapsdk/sysroots/armv7hf/usr/include/glib-2.0 -I/opt/axis/acapsdk/sysroots/armv7hf/usr/lib/glib-2.0/include -I/target-root/usr/include/opencv4 -L./lib -Wl,--no-as-needed,-rpath,'$ORIGIN/lib' -lm -lopencv_video -lopencv_imgproc -lopencv_core -lpthread -lvdostream -lsystemd -lgio-2.0 -lgobject-2.0 -lglib-2.0 imgprovider.cpp -o opencv_app && arm-linux-gnueabihf-strip --strip-unneeded opencv_app
#16 1.925 /usr/lib/gcc-cross/arm-linux-gnueabihf/9/../../../../arm-linux-gnueabihf/bin/ld: warning: libopencv_calib3d.so.4.5, needed by ./lib/libopencv_video.so, not found (try using -rpath or -rpath-link)
#16 1.925 /usr/lib/gcc-cross/arm-linux-gnueabihf/9/../../../../arm-linux-gnueabihf/bin/ld: /tmp/cclv1ky1.o: in function `main':
#16 1.925 /opt/app/example.cpp:101: undefined reference to `cv::SimpleBlobDetector::Params::Params()'
#16 1.925 /usr/lib/gcc-cross/arm-linux-gnueabihf/9/../../../../arm-linux-gnueabihf/bin/ld: /opt/app/example.cpp:136: undefined reference to `cv::SimpleBlobDetector::create(cv::SimpleBlobDetector::Params const&)'
#16 1.925 /usr/lib/gcc-cross/arm-linux-gnueabihf/9/../../../../arm-linux-gnueabihf/bin/ld: ./lib/libopencv_video.so: undefined reference to `cv::estimateAffine2D(cv::_InputArray const&, cv::_InputArray const&, cv::_OutputArray const&, int, double, unsigned int, double, unsigned int)'
#16 1.925 /usr/lib/gcc-cross/arm-linux-gnueabihf/9/../../../../arm-linux-gnueabihf/bin/ld: ./lib/libopencv_video.so: undefined reference to `cv::estimateAffinePartial2D(cv::_InputArray const&, cv::_InputArray const&, cv::_OutputArray const&, int, double, unsigned int, double, unsigned int)'
#16 1.925 collect2: error: ld returned 1 exit status
#16 1.926 make: *** [Makefile:30: opencv_app] Error 1
#16 1.927 
#16 1.927 make failed. Please fix above errors, before you can create a package
------
executor failed running [/bin/bash -o pipefail -c . /opt/axis/acapsdk/environment-setup* && acap-build .]: exit code: 1
stephanschulz@Komputer using-opencv % 
```

Corallo, 2022-05-09:
Hello! To build Opencv with more libraries you can do the following:

First, add the libraries to the build in the docker file, [line 76](https://github.com/AxisCommunications/acap-native-sdk-examples/blob/master/using-opencv/Dockerfile#L76):
` -D BUILD_LIST=core,imgproc,video,features2d,objdetect`

then in the makefile [Line 19](https://github.com/AxisCommunications/acap-native-sdk-examples/blob/master/using-opencv/app/Makefile#L19), you add those shared libs (and their dependency) 

`SHLIBS += -lm  -lopencv_calib3d -lopencv_features2d -lopencv_flann  -lopencv_video -lopencv_imgproc -lopencv_core -lpthread`

Then, you need to copy the .so files to the local lib directory, which means adding a new command in [line 21](https://github.com/AxisCommunications/acap-native-sdk-examples/blob/master/using-opencv/app/Makefile#L21) and [line 34](https://github.com/AxisCommunications/acap-native-sdk-examples/blob/master/using-opencv/app/Makefile#L34) to find and copy the right .so files.
To make this last step more simple, you could also copy all the opencv libs, 

replacing line 21,22,23 with:
`LIB_FILES = $(wildcard $(SDK_TARGET_LIBS)/libopencv_*.so*)`
and
34,35,36 with: 
`cp $(LIB_FILES) $(SHLIB_DIR)`

closed 2022-05-30 by pataxis