AxisCommunications/acap-native-sdk-examples #48: Unable to install eap using container example for large eap file size

garaujo23, 2022-05-05:
### Describe the bug

I am able to use the container example and install the sample eap on a camera and run the app. When I build a container using the computer vision sdk, I get error cannot install app and the logs show the following:
`2022-05-04T16:51:14.589+10:00 axis-b8a44f27fb6e [ WARNING ] packagemanager.cgi: pm_request_parser.c:513 pm_request_parser_init: Content-length 192176444 exceeds maximum length 104857600`

The content length matches my eap file size of 192Mb.

### To reproduce

1) Clone computervision sdk and go to [ml-minimal-example](https://github.com/AxisCommunications/acap-computer-vision-sdk-examples/tree/master/minimal-ml-inference)
2) Follow the steps to setup the export names for env and build the dockers for inference and model.
3) Save the dockers using docker save to .tar file
4) Copy the tar, docker compose, env files to the native sdk container example.
5) Update postinstall and preunsintall scripts to have your .tar file names
6) Build eap and copy as per the example.
7) Try and install the eap on the camera (should be 192Mb)

### Environment

- Axis device model: P3265-LVE
- Axis device firmware version:10.10.73
- OS and version: Ubuntu 18.04 LTS
- Version:SDK version 1.2, ACAP 4

marbali8, 2022-05-06:
Thank you for reaching out to us, we'll look in to this and get back to you.

garaujo23, 2022-05-13:
Hey @marbali8, any update on this?

marbali8, 2022-05-13:
Hey Gabriel,
@shreyasaxis is working on it. We will update you as soon as we have a solution or relevant information on the problem.

shreyasaxis, 2022-05-13:
Hey Gabriel, thanks for bringing up this issue with us. We have, indeed, reproduced the issue. 

We suggest you a workaround for this with help of upload cgi in combination with curl to send a file. The command could look something like this:

`curl -u <user:pass> -F"file=@<APP_FILE_PATH>" <DEVICE_IP>/axis-cgi/applications/upload.cgi`

Let us know if this works!

garaujo23, 2022-05-16:
Thanks @shreyasaxis I was able to install using postman. Longer term, the plan would be for customers to install the eap via the web UI, hard to ask them to do a terminal command to install. Is this max size going to change in the future?

shreyasaxis, 2022-05-16:
Yes, the limit causing this issue is found in AXIS OS but some more investigation on how to solve it will have to be done. We hope to have it fixed in a release coming soon.

shreyasaxis, 2022-05-16:
Add additional install command for container-example

Fixes  AxisCommunications/acap-native-sdk-examples#48
Reference: AFT-280

closed 2022-05-16 by shreyasaxis

garaujo23, 2022-06-06:
Hey @shreyasaxis wondering if there was any update on the size limit bug in installing eap through AXIS OS?

garaujo23, 2022-07-01:
Hey @marbali8 wondering if there was any update on the size limit bug in installing eap through AXIS OS?

pataxis, 2022-07-01:
Hi  @garaujo23 ,
I checked with the team and it's on the table, but there has not been time to start on it yet. The vacation period has started in Sweden and it's hard to say when it will be worked on, but I would not expect it to be fixed before 11.1 release at least.

pataxis, 2022-07-01:
I should also mention that the dimension of the issue is not fully known and more investigation is needed.

garaujo23, 2022-08-23:
Hey @pataxis any update on the upload size bug with eap? Wondering what to tell my customer after a few months now

pataxis, 2022-08-23:
Hi, the team has started looking at it and if it is a small fix it could get in to next release 11.1.

What is a typical setup for your customers? It sounds like these are smaller installations <20 devices if manual installation via web GUI is made. They are not using [Axis Device Manager](https://www.axis.com/support/tools/axis-device-manager) (ADM)?

garaujo23, 2022-08-23:
Great news, hopefully it is a small fix.

The customer has 100+ seperate sites with only 1-5 P3265-LVE per site to run the analytics. They have [NX witness](https://www.networkoptix.com/nx-witness) and planned to use NX cloud to deploy the app to sites via NX using the camera webpage. They don't have a single network across sites so Axis device manager would need to be installed at every sites server to use it, hence the web gui.