AxisCommunications/acap-native-sdk-examples #55: can't pass the compile

haichaoji1234, 2022-06-14:
when try to compile the 4.2 example, it report error：
#14 44.55   CC       libcurl_la-easy.lo
#14 44.63 In file included from easy.c:87:
#14 44.63 easy_lock.h: In function 'curl_simple_lock_lock':
#14 44.63 easy_lock.h:54:7: error: implicit declaration of function 'sched_yield' [-Werror=implicit-function-declaration]
#14 44.63    54 |       sched_yield();
#14 44.63       |       ^~~~~~~~~~~
#14 44.77 cc1: some warnings being treated as errors
#14 44.77 make[2]: *** [Makefile:2218: libcurl_la-easy.lo] Error 1
#14 44.77 make[2]: Leaving directory '/opt/build/curl/lib'
#14 44.77 make[1]: *** [Makefile:1355: all] Error 2
#14 44.77 make[1]: Leaving directory '/opt/build/curl/lib'
#14 44.77 make: *** [Makefile:1229: all-recursive] Error 1
------

I remembered that work with 4.0, but the example code keep unchanged. so it seems the curl version has some update, it can't work now. 

i also try to upload a local old curl version, and it can pass compile

haichaoji1234, 2022-06-14:
this also happens to acap3 example, so it seems curl new version involve this error.

pataxis, 2022-06-17:
@haichaoji1234  We have replaced the curl_example with [openssl_curl_example](https://github.com/AxisCommunications/acap-native-sdk-examples/tree/master/utility-libraries/openssl_curl_example) where the curl version is locked. I hope that example can get you through the issue.

closed 2022-06-17 by pataxis