AxisCommunications/acap-native-sdk-examples #61: Are tensorflow2 based trainings compatible with artpec8?

egSat, 2022-07-13:
Hi,
I'm having troubles when loading a custom model, the error says: 
ERROR in Inference: Failed to load model model.tflite (Could not load model: Could not build an interpreter of the model)
The model was trained with tensorflow2.

I converted the model to tflite and also tried converting it to google coral, but the issue remains. 
So my questions are:
-Is there any issue with using tensorflow2?
- What kind of optimizations we need to make when converting to tensorflow lite (for example input and/or outputs to int8, etc)?
- If a model is compatible with google coral, does it mean that will run on artpec8?

### Environment

- Axis device model: P3265
- Axis device firmware version: 10.11.73
- Version: sdk 1.2.1
- docker daemon: 1.2.3
### Additional context

Thank you very much

marbali8, 2022-07-14:
Hello!

Thank you for getting in touch with us. ARTPEC-8 cameras only support Tensorflow 1. There is a guide on how to train and convert your model from TensorFlow1 to Larod in the [tensorflow-to-larod-artpec8](https://github.com/AxisCommunications/acap-native-sdk-examples_staging/tree/master/tensorflow-to-larod-artpec8) example. Specifically, the model conversion is done in [convert_model.py](https://github.com/AxisCommunications/acap-native-sdk-examples_staging/blob/master/tensorflow-to-larod-artpec8/env/convert_model.py).

egSat, 2022-07-18:
Thank you very much. I'm trying to convert custom models trained from object-detection-api (tensorflow 1.15) , but I need to add more parameters to the conversor like --allow_custom_ops in order to work, but when loaded on the camera I get the same error (Could not load model: Could not build an interpreter of the model).

Corallo, 2022-07-27:
Hi, unfortunately custom ops are not supported at the moment, you can't use that model directly on the camera