AxisCommunications/acap-native-sdk-examples #66: Send detection results to local server.

QLinhHub, 2022-07-26:
Hello Axis ACAP developers,

Thanks to your code examples, I managed to train and integrate a car detection model (SSD MobilenetV2) on the Axis P1375 camera. It is very clear what you have described in these examples!
However, I still want to take a step forward in my project. I wonder how to send the detection results I have in my log file (on the Axis camera) to my local server (my laptop for example)? 
By the way, I tried ONVIF following your axevent example but I found out that it was not compatible in my case (sending bounding box coordinates).

So, if you have any suggestion for my problem, I would be very grateful!

Thanks in advance!

pataxis, 2022-07-27:
Hi @QLinhHub ,

yes the [send_event](https://github.com/AxisCommunications/acap-native-sdk-examples/tree/master/axevent/send_event) example shows how to declare your own [custom event](https://github.com/AxisCommunications/acap-native-sdk-examples/blob/master/axevent/send_event/app/send_event.c#L105) and sounds like an option for occasional detections.

I think that you would need to adapt the event declaration as well as the function sending the event to your data, probably updating `Value` field in the .c-file. Then of course also add some functionality that reads the detection results in the C-code, preferably directly instead of indirectly(like reading a log file). There is a helper script to help you confirm that your event got declared, see [axevent README](https://github.com/AxisCommunications/acap-native-sdk-examples/tree/master/axevent).

Hopefully the [axevent API documentation](https://axiscommunications.github.io/acap-documentation/docs/api/native-api.html#event-api) can be of some help in the adaptation.

Then it's a question to read the events from your server(laptop), could the [ONVIF documentation](https://www.onvif.org/specs/core/ONVIF-Core-Specification.pdf) chapter 9 be of any help to do this?

QLinhHub, 2022-07-28:
Hello @pataxis 

Thanks a lot for your answer!

I tried the [send_event](https://github.com/AxisCommunications/acap-native-sdk-examples/tree/master/axevent/send_event) example (I read the log file but of course I will change this approach in the future). In general, it works but there is the problem of the send time. 

For the [timer to declare event](https://github.com/AxisCommunications/acap-native-sdk-examples/blob/main/axevent/send_event/app/send_event.c#:~:text=app_data%2D%3Etimer%20%3D%20g_timeout_add_seconds(10%2C%20(GSourceFunc)send_event%2C%20app_data)%3B), the minimum value I can put here is 3 seconds. And for [time for fetching event], it is 4 seconds. So it takes some time (about 7 seconds) for each event to be sent. Do you think this is because I am reading the detection results indirectly from the log file? And if I can do it directly, do you think the latency will be significantly improved? 

Furthermore, the axevent library is designed to be used with the GLib library. An application using the axevent library must have a running GMainLoop (according to [axevent.h](https://axiscommunications.github.io/acap-documentation/docs/api/src/api/axevent/html/axevent_8h.html)), which makes integration with the object detection code (an infinite loop through the video frames) somewhat difficult.

I look forward to hearing your suggestions and ideas !

Thanks in advance!

Here is some detection results I got from the camera to my laptop (in a jupyter notebook):
[image]

pataxis, 2022-08-11:
Hi,
sounds great that it's up and running.

The API documentation says:

> A note on callback functions
> The callback functions registered with the AXEventHandler will be called from the GMainLoop thread in the default context. This means that the client may not prevent callback functions from returning, nor should any lengthy processing be made in the callback functions. Failure to comply with this convention will prevent the event system from, or delay it in, sending or delivering any more events to the calling application. 

Reading a log file sounds like it could be "lengthy processing" and I would expect that at least the 4 seconds should be possible to reduce if you move this handling directly to the C-code. This might also affect the callback time, i.e. the minimum 3 seconds you can set, if the event system gets blocked.

Another option to `axevent` could be to build a third party library like [mosquitto](https://mosquitto.org/) which sends `MQTT` events. This library has functions like [mosquitto_loop](https://mosquitto.org/api/files/mosquitto-h.html#mosquitto_loop) that works similar to `GMainLoop` in the [send_event](https://github.com/AxisCommunications/acap-native-sdk-examples/tree/main/axevent/send_event) example application.

QLinhHub, 2022-08-11:
Hello @pataxis 

Thank you very much for your response !

I managed to send an event from my camera to my laptop ! Thanks again !

Actually, I try to integrate yolov5s (tflite) into my camera and I get the error from larod: "setupLarod: Unable to load model: Could not allocate tensors". 

I haven't found the solution yet. But I think maybe the yolov5s is too big for my camera (ARTPEC-7 chip, 1GB RAM) or maybe there are some layers in the yolov5s that larod does not support ([larod custom layers](https://gitlab.com/unimatrix/larod/-/blob/master/doc/introduction-for-app-developers.md)).

Please let me know your ideas, it would be very helpful !

Thanks in advance !

pataxis, 2022-08-11:
Hi, great, then I will close this issue.

For the larod question it's better to open a new issue. It makes searching for Q/A easier for other developers.

closed 2022-08-11 by pataxis