AxisCommunications/acap-native-sdk-examples #70: Asynchronous connection has been closed

awais40, 2022-08-10:
I am getting error when i am upgrading my firmware from 10.9.4 to 10.11.76,

**setupLarod: Unable to load model: Could not load model: Asynchronous connection has been closed**

when I calling the function 

```
loadedModel = larodLoadModel(conn, larodModelFd, LAROD_ACCESS_PRIVATE,
                                 "object_detection", &error);
```

https://github.com/AxisCommunications/acap-native-sdk-examples/blob/main/object-detection/app/object_detection.c
line 335.

When I rollback to my previous firmware it start to work.

awais40, 2022-08-10:
I am using Axis P3265-LV Dome Camera for testing

awais40, 2022-08-10:
just found out it work fine when i am using -c 2 (CPU) just getting issue when i am using -c 12

pataxis, 2022-08-10:
Hi @awais40, thanks for reaching out.

Do you also get the error when installing the application directly on 10.11.76, not upgrading firmware?

awais40, 2022-08-10:
i dont get any error while installing the application. i get error after starting the application, when application tries to load the model file.

pataxis, 2022-08-11:
We will have a look at the problem and see if we can reproduce it in the coming days.

Do you use the [object-detection](https://github.com/AxisCommunications/acap-native-sdk-examples/tree/main/object-detection) example application without any modifications or do you have your own application?

marbali8, 2022-08-17:
Hello!

We are not able to reproduce the error on the firmware, camera and example you mentioned. Could you tell us more information on how you encountered the problem? For example, if you made changes on the object-detection example.

marbali8, 2022-08-22:
Closing due to inactivity.

closed 2022-08-22 by marbali8