AxisCommunications/acap-native-sdk-examples #71: setupLarod: Unable to load model: Could not load model: Could not allocate tensors

QLinhHub, 2022-08-11:
Hello AXIS ACAP developers,

Actually, I try to integrate yolov5s (tflite) into my camera and I get the error from larod: "setupLarod: Unable to load model: Could not allocate tensors".

I haven't found the solution yet. But I think maybe the yolov5s is too big for my camera (ARTPEC-7 chip, 1GB RAM) or maybe there are some layers in the yolov5s that larod does not support ([larod custom layers](https://gitlab.com/unimatrix/larod/-/blob/master/doc/introduction-for-app-developers.md)).

Please let me know your ideas, it would be very helpful !

Thanks in advance !

pataxis, 2022-08-11:
Hi @QLinhHub and thanks for the question.

We need you to provide more information.

- Which Axis product are you running on?
- I suppose you are building your own ACAP application (not one of the example applications), but which ACAP Native SDK version are you using? 
  - If you are basing your application on one of the example applications, that might also be helpful to know.
- When do you get the error?
  - Is it when building, installing or running the application?
  - Is the error output from the system log or are you running commands in a shell on the Axis device?
- Can you pin down which function call that gives this output?
  -  `setupLarod` is not a function in the [larod API documentation](https://axiscommunications.github.io/acap-documentation/docs/api/src/api/larod/html/index.html).
  - Are you using the same function `setupLarod` as in [vdo-larod.c](https://github.com/AxisCommunications/acap-native-sdk-examples/blob/main/vdo-larod/app/vdo_larod.c#L229) or any of the other examples?

QLinhHub, 2022-08-12:
Hi @pataxis,

Thanks for your response !

Here are the information you need:
- Camera: Axis P1375, chip ARTPEC-7, Memory: 1GB RAM and 512MB Flash, Firmware: 10.11.65, ACAP version: 4.0.
- I'm building my own object detection application base on your [object detection exemple](https://github.com/AxisCommunications/acap-native-sdk-examples/tree/main/object-detection). I'm trying to use [yolov5](https://github.com/ultralytics/yolov5) instead of SSD Mobilenetv2 like in your exemple because yolov5 is more powerful. 
- I tried yolov5s (small) and yolov5n (nano, much smaller) but I always face the same issue that Larod could not allocate tensors when loading the model [larodLoadModel](https://github.com/AxisCommunications/acap-native-sdk-examples/blob/main/object-detection/app/object_detection.c#:~:text=if%20(!loadedModel)%20%7B,%7D). The error occurs when I run the application, which mean that the build and installing phase is going well. Otherwise, the error message that I described appears on the system log. When I jump to the camera terminal and watch the log, that said: **[ NOTICE  ] systemd[1]: sdkobject_detection.service: Main process exited, code=exited, status=1/FAILURE**.
- The function [setupLarod](https://github.com/AxisCommunications/acap-native-sdk-examples/blob/main/object-detection/app/object_detection.c#:~:text=bool%20setupLarod(const,larodConn%2C%20larodModel**%20model)%20%7B) I used is in the object detection exemple.

I don't know if this is due to the camera capacity limit (because yolov5 input is 640x640, while for the Mobilenet SSD is 300x300) or larod constraints (maybe there are layers in yolov5 that larod doesn't support) or maybe I made some stupid codes ^^.

Thanks in advance !

Corallo, 2022-08-15:
@QLinhHub  Could you provide the model that you are trying to upload on the camera?

QLinhHub, 2022-08-16:
Hi @Corallo, yes you can download the model that I've already converted to .tflite format [here](https://drive.google.com/file/d/1bm1TYN4kmVpDP3gddDxsnnSYNlPLjjU7/view?usp=sharing). Just a note, because the input and output of this model is different from SSD MobilenetV2, so you have to make some modifications:
- The size of the input is 640x640.
- The output order of the model is: scores, locations, number of detections, classes.

Corallo, 2022-08-16:
Thanks, 
Did you update the input size in the [manifest](https://github.com/AxisCommunications/acap-native-sdk-examples/blob/main/object-detection/app/manifest.json.edgetpu#L13)?
It should go from 300 300 to 640 640.

After you run your example on the camera and it fails, can you ssh into the camera and run
`journalctl -u larod` and share the output

QLinhHub, 2022-08-16:
Thanks for your response! Here is the output of the command:

[image]

I think there is a problem with my tflite conversion. As a result, it produces some nodes in the graph that larod can't handle.

Corallo, 2022-08-16:
Right, so your camera does not support the full Tensorflow op set. And it seems that your model uses some that are not supported.
You can find the list of the supported ops [here](https://coral.ai/docs/edgetpu/models-intro/#supported-operations).

QLinhHub, 2022-08-17:
Hi, I think the problem is that I used [combined_non_max_removal](https://www.tensorflow.org/api_docs/python/tf/image/combined_non_max_suppression) which is not a TFLite operation but a TF operation. Therefore, it is not compatible. 

After doing some research, I realized that there are a few ways to include the TF operation by enabling the Flex delegate by linking a TensorFlow Lite Flex delegate shared library. Here are some useful sources: [ops_select](https://www.tensorflow.org/lite/guide/ops_select#cc), [github](https://github.com/tensorflow/tensorflow/blob/v2.3.0/tensorflow/lite/g3doc/guide/ops_select.md#c). 

Can you take a look at these and maybe provide a solution with cmake? This would be very useful as it would allow to install more types of models on the Axis camera! 

Thanks in advance!

Corallo, 2022-08-18:
Hi @QLinhHub 
If are you going to use the model only on cpu, the approach of rebuilding the full tensorflow in an acap should work. However, we can not provide assistance for it at the moment. 
If your camera supports the computer vision sdk, you could try installing it as a python library, which might be easier.