AxisCommunications/acap-native-sdk-examples #85: web-server example fails to build.

WilfR, 2022-10-19:
SDK version 1.4
last commit: 8a6fea5bebbd284a48b3b0b783c80bdc4ae96de9 ( Update versions (#83))

web-server example fails to build. 
To reproduce: 
1. "git pull" to get latest sources from acap-native-sdk-examples. 
2. stand in acap-native-sdk-examples/web-server
3. issue command `docker build . --tag axis-web-server:armv7hf`, which yields the following output : 
```
Sending build context to Docker daemon  35.84kB
Step 1/17 : ARG ARCH=armv7hf
Step 2/17 : ARG REPO=axisecp
Step 3/17 : ARG SDK=acap-native-sdk
Step 4/17 : ARG UBUNTU_VERSION=22.04
Step 5/17 : ARG VERSION=1.4
Step 6/17 : FROM ${REPO}/${SDK}:${VERSION}-${ARCH}-ubuntu${UBUNTU_VERSION}
 ---> a1a20c52894c
Step 7/17 : WORKDIR /opt
 ---> Using cache
 ---> 8bdab65f1dcb
Step 8/17 : COPY monkey.patch .
 ---> Using cache
 ---> 165c45ff2d43
Step 9/17 : RUN git clone -b v1.5.6 https://github.com/monkey/monkey
 ---> Using cache
 ---> 21e6c08f9f15
Step 10/17 : WORKDIR /opt/monkey
 ---> Using cache
 ---> af0c20e317cc
Step 11/17 : RUN git apply ../monkey.patch &&    . /opt/axis/acapsdk/environment-setup* &&    ./configure     --enable-shared     --malloc-libc     --prefix=/usr/local     --bindir=/usr/local/bin
  --libdir=/usr/local/lib     --sysconfdir=/usr/local/packages/monkey/html     --datadir=/usr/local/packages/monkey/html     --mandir=/usr/local/man     --logdir=/tmp     --plugdir=/usr/local/packag
es/monkey/lib     --pidfile=/tmp/monkey.pid     --incdir=/usr/local/include/monkey     --systemddir=/usr/lib/systemd/system &&    make &&    make install
 ---> Running in ad823b81e430
../monkey.patch:59: space before tab in indent.
        GMainLoop *loop;
../monkey.patch:60: space before tab in indent.
        int ret;
../monkey.patch:65: space before tab in indent.
        loop = g_main_loop_new(NULL, FALSE);
../monkey.patch:66: space before tab in indent.
        write_index();
../monkey.patch:95: space before tab in indent.
        g_main_loop_run(loop);
warning: configure has type 100755, expected 100644
error: patch failed: configure:1297
error: configure: patch does not apply
error: patch failed: examples/Makefile:1
error: examples/Makefile: patch does not apply
error: patch failed: examples/hello.c:1
error: examples/hello.c: patch does not apply
error: patch failed: examples/list.c:1
error: examples/list.c: patch does not apply
error: patch failed: examples/quiz.c:1
error: examples/quiz.c: patch does not apply
error: patch failed: src/mk_server.c:40
error: src/mk_server.c: patch does not apply
The command '/bin/sh -c git apply ../monkey.patch &&    . /opt/axis/acapsdk/environment-setup* &&    ./configure     --enable-shared     --malloc-libc     --prefix=/usr/local     --bindir=/usr/local
/bin     --libdir=/usr/local/lib     --sysconfdir=/usr/local/packages/monkey/html     --datadir=/usr/local/packages/monkey/html     --mandir=/usr/local/man     --logdir=/tmp     --plugdir=/usr/local
/packages/monkey/lib     --pidfile=/tmp/monkey.pid     --incdir=/usr/local/include/monkey     --systemddir=/usr/lib/systemd/system &&    make &&    make install' returned a non-zero code: 1
```

In particular observe that the docker build failed in attempting to execute the "git apply ../monkey.patch" command in Step 11. 
SDK version 1.2 of this same web-server example builds fine on the same host computer. 

Please also note that there is an error in the "How to build the code" section for the web-server example at https://github.com/AxisCommunications/acap-native-sdk-examples/tree/main/web-server. In this section the user is asked to run the command `docker build . --build-arg ARCH --tag web-server:$ARCH`. I believe that the correct command should be `docker build . --build-arg ARCH=$ARCH --tag web-server:$ARCH`.

pataxis, 2022-10-19:
Hi @WilfR, thanks for reaching out.

This application built without a problem on for example the standard `ubuntu-latest` container image in GitHub Actions.

- What is your OS, Docker version and build environment - are you building inside a Docker container or directly in your host OS?
- Have you modified the example or running it as it is?

WilfR, 2022-10-19:
Hi @pataxis - thanks for having a look at this issue. 

I am building on a Windows 11 host computer, right from the command line. I just open a command line interface (CLI) window, navigate to the `web-server` folder and give the `docker build . --tag axis-web-server:armv7hf` command. The results are exactly the same whether I try to build from a plain old Microsoft Windows `cmd` CLI or from an Ubuntu 20.04 Windows Subsystem for Linux terminal window. 

Here is my Docker version information:

```
F:\Home\Projects\_Active\acap-native-sdk-examples\web-server>docker version

Client:
 Cloud integration: v1.0.24
 Version:           20.10.17
 API version:       1.41
 Go version:        go1.17.11
 Git commit:        100c701
 Built:             Mon Jun  6 23:09:02 2022
 OS/Arch:           windows/amd64
 Context:           default
 Experimental:      true

Server: Docker Desktop 4.10.1 (82475)
 Engine:
  Version:          20.10.17
  API version:      1.41 (minimum version 1.12)
  Go version:       go1.17.11
  Git commit:       a89b842
  Built:            Mon Jun  6 23:01:23 2022
  OS/Arch:          linux/amd64
  Experimental:     false
 containerd:
  Version:          1.6.6
  GitCommit:        10c12954828e7c7c9b6e0ea9b0c02b01407d3ae1
 runc:
  Version:          1.1.2
  GitCommit:        v1.1.2-0-ga916309
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0
```

I have not modified the example in any way - I just now used `git status` and `git pull` to confirm that I have a complete and faithful reproduction of the current state of the repo. I originally got the repo from github onto my local computer with the command `git clone https://github.com/AxisCommunications/acap-native-sdk-examples.git`.

If I compare the "SDK Version 1.2" version of the web-server example (which builds on my machine) with the "SDK Version 1.4" version of the web-server example (which does not build on my machine) I see the only differences are in the Dockerfile:

In the SDK Version 1.2 the Dockerfile contains the lines
```
ARG UBUNTU_VERSION=20.04
ARG VERSION=1.2
```

In the SDK Version 1.4 the Dockerfile contains 
```
ARG UBUNTU_VERSION=22.04
ARG VERSION=1.4
```

The folders are otherwise identical (apart from a small change in the LICENSE).

WilfR, 2022-10-19:
For the sake of completeness I just tried installing a new Ubuntu 22.04 kernel in my Windows 11 computer and trying to build the web-server example from a WSL Ubuntu 22.04 terminal command line. However there was no change in results - the build still failed in Step 11 trying to execute `RUN git apply ../monkey.patch`.

pataxis, 2022-10-20:
Thanks for the update, we will investigate and get back.

> Please also note that there is an error in the "How to build the code" 

The command assumes that the variable is exported as an environment variable as is stated in the instructions.
```sh
# Set your device IP address and architecture
export ARCH=<armv7hf or aarch64>
export DEVICE_IP=<device IP address>
export PASS=<device password>

docker build . --build-arg ARCH --tag web-server:$ARCH
```

WilfR, 2022-10-20:
OK, thanks for the update on the investigation @pataxis. 

As far as the error in the "How to build the code" documentation, yes I understand that ARCH is an exported environment variable, but I believe there is still a problem with the documented example as written. 

Suppose we had defined the value of the ARCH variable to be `aarch64` so we had the following snippet:

```
export ARCH=aarch64
docker build . --build-arg ARCH --tag web-server:$ARCH
```

Then the `docker build` command gets expanded to 
```
docker build . --build-arg ARCH --tag web-server:aarch64
```
(the `$ARCH` in the `--tag` gets replaced by the value of the `ARCH` variable, but the `ARCH` string in the `--build-arg` is not replaced by its value because it does not start with `$`.)

This is not what we want because the syntax of the `--build-arg` option is `--build-arg <name>=<value>`. The correct `docker build` command we want to execute in this case is `docker build . --build-arg ARCH=aarch64 --tag web-server:aarch64`. We can achieve this with the snippet 
```
export ARCH=aarch64
docker build . --build-arg ARCH=$ARCH --tag web-server:$ARCH
```

Finally, the documented example will work by accident for the `armv7hf` case because the invalid `--build-arg ARCH` option will be ignored, and the Dockerfile assumes `armv7hf` is the default value for the `ARCH` argument within the Dockerfile. 

In summary I believe that 
```
docker build . --build-arg ARCH=$ARCH --tag web-server:$ARCH
```
is the correct replacement for the current text 
```
docker build . --build-arg ARCH --tag web-server:$ARCH
```
but perhaps I have overlooked something or made some unwarranted assumption.

jaryder, 2022-11-09:
It should work as-is according to Docker --build-arg documentation:
https://docs.docker.com/engine/reference/commandline/build/#set-build-time-variables---build-arg
Seen the example:
`export HTTP_PROXY=http://10.20.30.2:1234
docker build --build-arg HTTP_PROXY .`

I'm surprised your first attempt didn't work (omitting ARCH) since it is already defined in the Dockerfile.  Glad to hear it is building successfully for you now :).