AxisCommunications/acap-native-sdk-examples #111: get VdoBuffer too slow

vuhailongkl97, 2023-02-10:
### Describe the bug
currently I use your example to get vdoBuffer  in using-opencv example (this repo), I got 1-2 fps. 
I use printf with "\n" to force log to screen instead of syslog it's still slow
### To reproduce
use example using-opencv to build an eap application and load to Axis device Q1615 MKIII

Steps to reproduce the behavior:

1. Set up clone this repo
2. Do this make using-opencv .eap app and load to the device on its web
3. check log and see it very slow, 

### Screenshots

### Environment

- Axis device model:Q1615 Mk III
- Axis device firmware version:11.2.61
- Stack trace or logs: 
- 2023-02-10T12:53:37.448+09:00 axis-b8a44f3a8b10 [ INFO    ] opencv_app[979]: Last message 'Motion detected: YES' repeated 6 times, suppressed by syslog-ng on axis-b8a44f3a8b10
2023-02-10T12:53:37.543+09:00 axis-b8a44f3a8b10 [ INFO    ] opencv_app[16994]: Motion detected: YES
2023-02-10T12:53:38.448+09:00 axis-b8a44f3a8b10 [ INFO    ] opencv_app[979]: Last message 'Motion detected: YES' repeated 6 times, suppressed by syslog-ng on axis-b8a44f3a8b10
2023-02-10T12:53:38.495+09:00 axis-b8a44f3a8b10 [ INFO    ] opencv_app[16994]: Motion detected: YES
2023-02-10T12:53:39.448+09:00 axis-b8a44f3a8b10 [ INFO    ] opencv_app[979]: Last message 'Motion detected: YES' repeated 6 times, suppressed by syslog-ng on axis-b8a44f3a8b10
2023-02-10T12:53:39.469+09:00 axis-b8a44f3a8b10 [ INFO    ] opencv_app[16994]: Motion detected: YES
2023-02-10T12:53:40.448+09:00 axis-b8a44f3a8b10 [ INFO    ] opencv_app[979]: Last message 'Motion detected: YES' repeated 7 times, suppressed by syslog-ng on axis-b8a44f3a8b10
2023-02-10T12:53:40.574+09:00 axis-b8a44f3a8b10 [ INFO    ] opencv_app[16994]: Motion detected: YES
2023-02-10T12:53:41.448+09:00 axis-b8a44f3a8b10 [ INFO    ] opencv_app[979]: Last message 'Motion detected: YES' repeated 6 times, suppressed by syslog-ng on axis-b8a44f3a8b10
2023-02-10T12:53:41.543+09:00 axis-b8a44f3a8b10 [ INFO    ] opencv_app[16994]: Motion detected: YES
2023-02-10T12:53:42.448+09:00 axis-b8a44f3a8b10 [ INFO    ] opencv_app[979]: Last message 'Motion detected: YES' repeated 6 times, suppressed by syslog-ng on axis-b8a44f3a8b10
2023-02-10T12:53:42.509+09:00 axis-b8a44f3a8b10 [ INFO    ] opencv_app[16994]: Motion detected: YES
2023-02-10T12:53:43.448+09:00 axis-b8a44f3a8b10 [ INFO    ] opencv_app[979]: Last message 'Motion detected: YES' repeated 6 times, suppressed by syslog-ng on axis-b8a44f3a8b10
2023-02-10T12:53:43.475+09:00 axis-b8a44f3a8b10 [ INFO    ] opencv_app[16994]: Motion detected: YES
2023-02-10T12:53:44.448+09:00 axis-b8a44f3a8b10 [ INFO    ] opencv_app[979]: Last message 'Motion detected: YES' repeated 7 times, suppressed by syslog-ng on axis-b8a44f3a8b10
2023-02-10T12:53:44.572+09:00 axis-b8a44f3a8b10 [ INFO    ] opencv_app[16994]: Motion detected: YES
2023-02-10T12:53:45.448+09:00 axis-b8a44f3a8b10 [ INFO    ] opencv_app[979]: Last message 'Motion detected: YES' repeated 6 times, suppressed by syslog-ng on axis-b8a44f3a8b10
- Version:  used docker to build the eap application

pataxis, 2023-02-10:
Hi @vuhailongkl97 , thank you for reaching out. We'll get back to you when we have had time to look in to your issue.

Corallo, 2023-02-15:
Hello @vuhailongkl97 

The application goes faster than 1fps, but as you see in your log, since the output is always the same, it is suppressed by the system log. 
e.g. 2023-02-10T12:53:37.448+09:00 axis-b8a44f3a8b10 [ INFO ] opencv_app[979]: Last message 'Motion detected: YES' repeated 6 times, suppressed by syslog-ng on axis-b8a44f3a8b10

By only capturing the frames, you should get performances that are close to the camera frame rate settings (30 or 60 fps).
of course, if you process the frame on top of it, that will slow down your throughput. In the specific case of this application, I think 
` bgsub->apply(bgr_mat, fg, 0.005);` is the responsible for the slowdown from 30 to 7-8 fps. You can double-check by adding a unique number to each print to avoid suppression.

On artpect-8 devices, the performances should be 2-3 times better.

vuhailongkl97, 2023-02-15:
@Corallo Thank you, I got it.

closed 2023-02-15 by vuhailongkl97