AxisCommunications/acap-native-sdk-examples #118: Kernel crashed

awais40, 2023-03-15:
I am working on Camera AXIS P3265-LV. I had reported a bug on firmware 10, when i call the function 

**loadedModel = larodLoadModel(conn, larodModelFd, LAROD_ACCESS_PRIVATE, "object_detection", &error);**
 
i got error message  **Asynchronous connection has been closed**

The issue was resolved in firmware 11. It worked fine on firmware 11.0.108 and 11.1.72 but on the latest release 11.3.57, it start to restart my camera when the same above command is called. When i downgrade to 11.1.72, it again start to work fine
following are the logs ( i have tested on 3 different cameras)

[Crash.txt](https://github.com/AxisCommunications/acap-native-sdk-examples/files/10978660/Crash.txt)

[image]

Corallo, 2023-03-15:
Hello @awais40 

Are the three different cameras that you tested all P3265-LV?
I guess you are not using the model of the example, but rather your own right?
What I can say without having the model is that this is probably an out of memory error.

It is possible that firmware updates comes with changes in the loading model function, or the drivers beneath it, that takes slightly more or slightly less memory. 
It might be that because of your model size, the memory consumption to load it is right on the edge of available RAM of your device.
You could try to solve it using a smaller model, or a device with more RAM, or make sure that the device is not running other acaps that consume memory.

If you would like us to check more to this issue we would need to take a look at the model to see if we can replicate it, you could send an untrained version of it if you prefer.

awais40, 2023-03-15:
@Corallo 
Yes, all of them were P3265 and Yes we are using our own custom trained model. 
There might be some memory leakage in the new firmware that load the model? as it works fine on previous firmware version.

Corallo, 2023-03-15:
It is possible, but could also be that the loading in this new firmware consume "non-leaking" more memory, we have seen it before.
On our tests, we didn't notice any leak so far.

closed 2023-04-04 by pataxis