AxisCommunications/acap-native-sdk-examples #138: Black cropped image is output when detect object in object detection

takuya-takeuchi, 2023-07-14:
### Describe the bug

Black cropped image is output when detect object in object detection.
Camera image is clearly and no problem.

### To reproduce

Run [object-detection](https://github.com/AxisCommunications/acap-native-sdk-examples/tree/main/object-detection) in AXIS Q1656 Box Camera anc check /tmp/*/jpg images.

### Screenshots

N/A

### Environment

- Axis device model: [AXIS Q1656 Box Camera]
- Axis device firmware version: [10.12]
- Stack trace or logs: [N/A]
- OS and version: [Ubuntu 18.04 LTS]
- Version: [N/A]

### Additional context

* Yuv image is not black but `cropAddr` is black image

pataxis, 2023-07-14:
Hi @takuya-takeuchi , thanks for your report, we'll get back to you when we had time to look at this issue.

Corallo, 2023-07-18:
Hi @takuya-takeuchi 

Before we dig into this issue, it would be good if you can try with a newer firmware.
Between 10.12 and 11.0 there have been some breaking changes, and the examples are meant to work on the latest firmware.

Search here:
https://www.axis.com/support/firmware
For the firmware for your Q1656 and try with the latest (should be 11.5).

takuya-takeuchi, 2023-07-19:
@Corallo 
Thank you for your quick response and advice.
I updated firmeware to 11.5.64 and rebuild app on axisecp/acap-native-sdk:1.9-aarch64-ubuntu22.04.
But problem is not resolved.

This is object detection's log.

````
----- Contents of SYSTEM_LOG for 'object_detection' -----

2023-07-19T08:42:53.671+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Starting /usr/local/packages/object_detection/object_detection
2023-07-19T08:42:54.005+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: 'buffer.strategy': <uint32 3>
2023-07-19T08:42:54.005+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: 'channel'--------: <uint32 1>
2023-07-19T08:42:54.005+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: 'format'---------: <uint32 3>
2023-07-19T08:42:54.005+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: 'height'---------: <uint32 300>
2023-07-19T08:42:54.005+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: 'width'----------: <uint32 480>
2023-07-19T08:42:54.005+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Creating VDO image provider and creating stream 480 x 300
2023-07-19T08:42:54.005+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Dump of vdo stream settings map =====
2023-07-19T08:42:54.005+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: chooseStreamResolution: We select stream w/h=480 x 300 based on VDO channel info.
2023-07-19T08:42:54.348+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: 'buffer.strategy': <uint32 3>
2023-07-19T08:42:54.348+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: 'channel'--------: <uint32 1>
2023-07-19T08:42:54.348+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: 'format'---------: <uint32 3>
2023-07-19T08:42:54.348+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: 'height'---------: <uint32 1080>
2023-07-19T08:42:54.348+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: 'width'----------: <uint32 1920>
2023-07-19T08:42:54.348+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Creating VDO raw image provider and stream 1920 x 1080
2023-07-19T08:42:54.348+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Dump of vdo stream settings map =====
2023-07-19T08:42:54.695+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Calculate crop image
2023-07-19T08:42:54.695+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Create larod models
2023-07-19T08:42:54.695+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Create preprocessing maps
2023-07-19T08:42:54.695+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Crop VDO image X=90 Y=0 (300 x 300)
2023-07-19T08:42:54.695+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Setting up larod connection with chip axis-a8-dlpu-tflite, model /usr/local/packages/object_detection/model/converted_model.tflite and label file /usr/local/packages/object_detection/label/labels.txt
2023-07-19T08:42:54.746+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Available chip IDs:
2023-07-19T08:42:54.746+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Chip: axis-a8-dlpu-native
2023-07-19T08:42:54.746+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Chip: axis-a8-dlpu-proc
2023-07-19T08:42:54.746+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Chip: axis-a8-dlpu-tflite
2023-07-19T08:42:54.746+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Chip: axis-a8-gpu-proc
2023-07-19T08:42:54.746+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Chip: axis-ace-proc
2023-07-19T08:42:54.746+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Chip: cpu-proc
2023-07-19T08:42:54.746+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Chip: cpu-tflite
2023-07-19T08:43:29.091+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Allocate memory for input/output buffers
2023-07-19T08:43:29.091+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Create input/output tensors
2023-07-19T08:43:29.091+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Determine tensor buffer sizes
2023-07-19T08:43:29.091+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: Loading preprocessing model with chip cpu-proc
2023-07-19T08:43:29.091+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: createAndMapTmpFile: Setting up a temp fd with pattern /tmp/crop.test-XXXXXX and size 6220800
2023-07-19T08:43:29.091+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: createAndMapTmpFile: Setting up a temp fd with pattern /tmp/larod.in.test-XXXXXX and size 270000
2023-07-19T08:43:29.091+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: createAndMapTmpFile: Setting up a temp fd with pattern /tmp/larod.out1.test-XXXXXX and size 320
2023-07-19T08:43:29.091+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: createAndMapTmpFile: Setting up a temp fd with pattern /tmp/larod.out2.test-XXXXXX and size 80
2023-07-19T08:43:29.091+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: createAndMapTmpFile: Setting up a temp fd with pattern /tmp/larod.out3.test-XXXXXX and size 80
2023-07-19T08:43:29.091+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: createAndMapTmpFile: Setting up a temp fd with pattern /tmp/larod.pp.test-XXXXXX and size 216000
[... cut ...]
2023-07-19T08:43:29.602+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: 4: Object 1: Classes: person - Scores: 0.558594 - Locations: [0.397288,0.001052,0.910682,0.267301] [2,429,511,554]
2023-07-19T08:43:29.602+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: free
2023-07-19T08:43:29.602+09:00 axis-b8a44f7b9f0b [ INFO    ] object_detection[8172]: jpeg_to_file
````

This has some moification to output yuv (480x300) and original image (1920x1080) but it should not affect to this issue.
yuv is correct but original and cropped images are blacks.

takuya-takeuchi, 2023-07-19:
FYI
I just tried original object detection but detected image is still black.
A revison of source code is dbb91e94b5f508fb63e3830944044002d65e5014.

takuya-takeuchi, 2023-07-21:
@Corallo 
Hi, do you have any information to resolve issue?
Thank you for your cooperations.

marbali8, 2023-07-24:
Hello,
we will look into the issue as soon as we can and let you know :)

Corallo, 2023-08-30:
Hello @takuya-takeuchi 
It has been a while, but we are working on it. We are going to publish a fix probably this week.

Corallo, 2023-09-01:
Hello @takuya-takeuchi 
We need to make more tests before merging the fix as we noticed other problems,
In the meantime, I created a [branch](https://github.com/AxisCommunications/acap-native-sdk-examples/tree/object_detector_fix) that you can try out. it should have the issue with the black cropped image solved.

closed 2023-09-08 by Corallo

pataxis, 2023-09-11:
Hi @takuya-takeuchi , the problem should have been fixed and is now on main branch. Thanks for reporting!

takuya-takeuchi, 2023-09-12:
I confirmed issue is gone away.
Thank you for your kindly support

commit a844dd17 (2023-09-11): Fix bug that creates an empty output image

Bug in object-detection* examples.
Fixes #138
PR #148 (2023-09-11): Fix bug that creates an empty output image
Bug in object-detection* examples.
Fixes #138
commit 2349aa57 (2023-09-11): Fix bug that creates an empty output image

Bug in object-detection* examples.
Fixes #138