AxisCommunications/acap-native-sdk-examples #141: Inference is performed, but no object is detected.(object_detection-cv25)

tosamu51, 2023-09-04:
### Describe the bug

Inference is performed, but no object is detected.(object_detection-cv25)

### To reproduce

Run object-detection-cv25 in M2035-LE and M3085-V ,Inference is performed, but no object is detected.

object_detection :detect object Person,Chair...   (M2035_ObjectDetect.txt)
			https://github.com/AxisCommunications/acap-native-sdk-examples/tree/main/object-detection

 object_detection-cv25 :Not detect object      (M2035_ObjectDetectCV25.txt)
			https://github.com/AxisCommunications/acap-native-sdk-examples/tree/main/object-detection-cv25

Camera1：AXIS M2035-LE（11.5.64)
Camera2：AXIS M3085-V（11.5.64)
  AXIS M2035-LE and AXIS M3085-V（11.5.64) is same phenomenon

### Log

2023-09-04T04:38:54.927+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Starting /usr/local/packages/object_detection/object_detection
2023-09-04T04:38:55.258+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: 'buffer.strategy': <uint32 3>
2023-09-04T04:38:55.258+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: 'channel'--------: <uint32 1>
2023-09-04T04:38:55.258+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: 'format'---------: <uint32 3>
2023-09-04T04:38:55.258+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: 'height'---------: <uint32 720>
2023-09-04T04:38:55.258+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: 'width'----------: <uint32 1280>
2023-09-04T04:38:55.258+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Creating VDO image provider and creating stream 1280 x 720
2023-09-04T04:38:55.258+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Dump of vdo stream settings map =====
2023-09-04T04:38:55.258+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: chooseStreamResolution: We select stream w/h=1280 x 720 based on VDO channel info.
2023-09-04T04:38:55.644+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: 'buffer.strategy': <uint32 3>
2023-09-04T04:38:55.644+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: 'channel'--------: <uint32 1>
2023-09-04T04:38:55.644+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: 'format'---------: <uint32 3>
2023-09-04T04:38:55.644+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: 'height'---------: <uint32 360>
2023-09-04T04:38:55.644+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: 'width'----------: <uint32 640>
2023-09-04T04:38:55.644+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Creating VDO raw image provider and stream 640 x 360
2023-09-04T04:38:55.644+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Dump of vdo stream settings map =====
2023-09-04T04:38:55.820+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: raw width=640, raw height=360 
2023-09-04T04:38:55.879+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Calculate crop image
2023-09-04T04:38:55.879+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Create larod models
2023-09-04T04:38:55.879+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Create preprocessing maps
2023-09-04T04:38:55.879+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Creating VDO crop image provider and creating stream 1280 x 720
2023-09-04T04:38:55.879+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Crop VDO image X=280 Y=0 (720 x 720)
2023-09-04T04:38:55.879+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Setting up larod connection with chip ambarella-cvflow, model /usr/local/packages/object_detection/model/converted_model.bin and label file /usr/local/packages/object_detection/label/labels.txt
2023-09-04T04:38:55.879+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: chooseStreamResolution: We select stream w/h=1280 x 720 based on VDO channel info.
2023-09-04T04:38:55.911+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Available chip IDs:
2023-09-04T04:38:55.911+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Chip: ambarella-cvflow
2023-09-04T04:38:55.911+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Chip: ambarella-cvflow-proc
2023-09-04T04:38:55.911+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Chip: cpu-proc
2023-09-04T04:38:55.911+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Chip: cpu-tflite
2023-09-04T04:38:56.110+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Allocate memory for input/output buffers
2023-09-04T04:38:56.110+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Create input/output tensors
2023-09-04T04:38:56.110+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Creating temporary files and memmaps for inference input and output tensors
2023-09-04T04:38:56.110+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Determine tensor buffer sizes
2023-09-04T04:38:56.110+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Loading preprocessing model with chip cpu-proc
2023-09-04T04:38:56.110+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: createAndMapTmpFile: Setting up a temp fd with pattern /tmp/crop.test-XXXXXX and size 691200
2023-09-04T04:38:56.110+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: createAndMapTmpFile: Setting up a temp fd with pattern /tmp/larod.in.test-XXXXXX and size 288000
2023-09-04T04:38:56.110+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: createAndMapTmpFile: Setting up a temp fd with pattern /tmp/larod.out1.test-XXXXXX and size 30672
2023-09-04T04:38:56.110+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: createAndMapTmpFile: Setting up a temp fd with pattern /tmp/larod.out2.test-XXXXXX and size 697788
2023-09-04T04:38:56.110+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: createAndMapTmpFile: Setting up a temp fd with pattern /tmp/larod.pp.test-XXXXXX and size 1382400
2023-09-04T04:38:56.114+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Connect tensors to file descriptors
2023-09-04T04:38:56.114+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Create job requests
2023-09-04T04:38:56.114+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Found 1 input tensors and 2 output tensors
2023-09-04T04:38:56.114+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Set input tensors
2023-09-04T04:38:56.114+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Set output tensors
2023-09-04T04:38:56.114+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Start fetching video frames from VDO
2023-09-04T04:38:56.128+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Converted image in 22 ms
2023-09-04T04:38:56.157+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Ran inference for 27 ms
2023-09-04T04:38:56.257+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Postprocesing in 101 ms
2023-09-04T04:38:56.271+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Converted image in 13 ms
2023-09-04T04:38:56.298+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Ran inference for 26 ms
[... 5 lines cut ...]
2023-09-04T04:38:56.586+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Ran inference for 26 ms
2023-09-04T04:38:56.686+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Postprocesing in 99 ms
2023-09-04T04:38:56.699+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Converted image in 13 ms
2023-09-04T04:38:56.726+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Ran inference for 26 ms
2023-09-04T04:38:56.828+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Postprocesing in 101 ms
2023-09-04T04:38:56.842+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Converted image in 14 ms
2023-09-04T04:38:56.870+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Ran inference for 27 ms
2023-09-04T04:38:56.969+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Postprocesing in 99 ms
2023-09-04T04:38:56.986+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Converted image in 16 ms
2023-09-04T04:38:57.014+00:00 axis-b8a44f45d2c1 [ INFO    ] object_detection[1292144]: Ran inference for 28 ms
[... cut ...]
- Axis device firmware version: [11.5.64)]
- Stack trace or logs: [N/A]
- OS and version: [Ubuntu 22.04 LTS]
- Additional context

[M2035_ObjectDetectCV25.txt](https://github.com/AxisCommunications/acap-native-sdk-examples/files/12509662/M2035_ObjectDetectCV25.txt)
[M3085_ObjectDtectCV25.txt](https://github.com/AxisCommunications/acap-native-sdk-examples/files/12509664/M3085_ObjectDtectCV25.txt)
[M2035_ObjectDetect.txt](https://github.com/AxisCommunications/acap-native-sdk-examples/files/12509665/M2035_ObjectDetect.txt)

pataxis, 2023-09-12:
Hi @tosamu51 , this issue should have been fixed and has been synced to main branch. Could you retest?

tosamu51, 2023-09-13:
Thanks for the fix.
We have confirmed that it works with AXIS M2035-LE (11.5.64).

[object_detectionLOG.txt](https://github.com/AxisCommunications/acap-native-sdk-examples/files/12592829/object_detectionLOG.txt)

closed 2023-09-13 by pataxis