AxisCommunications/acap-native-sdk-examples #154: Q6215-LE Supported?

whittenator, 2023-09-12:
Is the Q6215-LE supported to run Docker ACAP? 

I followed the compatibility section here: https://github.com/AxisCommunications/docker-compose-acap

Ran the following command: 
`
ssh root@$DEVICE_IP 'command -v containerd >/dev/null 2>&1 && echo Compatible with Docker ACAP || echo Not compatible with Docker ACAP'
`

But it says: "Not compatible with Docker ACAP". I was wanting to run Object Detection on the camera, is that not possible for this model?

whittenator, 2023-09-12:
I attempted to install the 1.4.1 .eap[ docker ](https://github.com/AxisCommunications/docker-acap/releases/tag/1.4.1) as an app on the Q6215-LE through the Web UI. However, it fails to start and I get this in the systems log:
```
2023-09-12T16:04:51.234-04:00 axis-b8a44f212395 [ ERR     ] acapctl: code 1: ../git/src/util.c:525: Failed running "systemctl restart --quiet 'sdkdockerdwrapper.service'"
2023-09-12T16:04:51.272-04:00 axis-b8a44f212395 [ WARNING ] acapmanager: acapmanager-dbus.c:1561 Failed to start dockerdwrapper
2023-09-12T16:04:51.302-04:00 axis-b8a44f212395 [ WARNING ] acapmanager: acapmanager-dbus.c:386 acapctl error 1: Internal error, contact maintainer
```

pataxis, 2023-09-13:
Hi @whittenator , thanks for your question. 

As the compatibility test responds, Q6215-LE which is an Artpec-6 product does not have container support.

You can find a full list of products with container and DLPU support in [ACAP 4 documentation](https://axiscommunications.github.io/acap-documentation/docs/axis-devices-and-compatibility/#acap-computer-vision-sdk).

whittenator, 2023-09-13:
> Hi @whittenator , thanks for your question.
>
>
>
> As the compatibility test responds, Q6215-LE which is an Artpec-6 product does not have container support.
>
>
>
> You can find a full list of products [...]

Thanks for the speedy reply! Looks like I will have to use the Native SDK to develop a plugin. Does the Native SDK give the ability to listen to focus and send focus updates?

pataxis, 2023-09-14:
I'm not sure if there is. You could have a look at the [axevent](https://github.com/AxisCommunications/acap-native-sdk-examples/tree/main/axevent) examples and see if there is any event that suits your needs by [retrieving the eventlist](https://github.com/AxisCommunications/acap-native-sdk-examples/tree/main/axevent#find-events-using-wrapper).

whittenator, 2023-09-14:
> I'm not sure if there is. You could have a look at the [axevent](https://github.com/AxisCommunications/acap-native-sdk-examples/tree/main/axevent) examples and see if there is any event that suits [...]

@pataxis You can control the camera's PTZ(Pan, Tilt, Zoom) and focus through the events using that API? I thought the axevent was just for subscribing to events such as "camera is moving" is such.

pataxis, 2023-09-15:
I thought there might be an event for focus change in axevent to listen to but I'm not sure of this.
On the other hand, I don't know any way today to set the focus from the ACAP application.
The VAPIX API for [Optics Control](https://www.axis.com/vapix-library/subjects/t10175981/section/t10168762/display) has methods to handle focus.

pataxis, 2023-09-20:
Will close this issue since the original question was answered and the #155  was opened.

closed 2023-09-20 by pataxis