AxisCommunications/acap-native-sdk-examples #156: How to train dataset for object_detection

takuya-takeuchi, 2023-09-13:
### Describe the feature

How to train object detection model?
There is no information in this repo.

### Added value

https://github.com/AxisCommunications/acap-native-sdk-examples/tree/main/object-detection uses models of https://github.com/google-coral/test_data so I think model would be generated from https://github.com/tensorflow/models
But I'm not sure it is correct. 
In convlustion, I want to know 

* training script and repo
  * git revision or tag (if use repo)
* training step
  * and convert script for trained model to use on axis camera

### Additional context

N/A

shreyasatwork, 2023-09-14:
Hello,

Thanks for your question!

Although we don't specifically provide training scripts for object detection models, we do it for an object classification model in [tensorflow-to-larod](https://github.com/AxisCommunications/acap-native-sdk-examples/tree/main/tensorflow-to-larod), [tensorflow-to-larod-artpec8](https://github.com/AxisCommunications/acap-native-sdk-examples/tree/main/tensorflow-to-larod-artpec8), [tensorflow-to-larod-cv25](https://github.com/AxisCommunications/acap-native-sdk-examples/tree/main/tensorflow-to-larod-cv25), they also provide the conversion script to convert the resulting model to tflite format.

takuya-takeuchi, 2023-09-14:
@shreyasatwork 
thanks quick response.

After read above code and README, I have some question about TF.

On https://github.com/AxisCommunications/acap-native-sdk-examples/issues/61, it is said that TF2 is not supported.
TF1 is too old. Most of current models targets modern framework. They would support conversion model except for TF1. Is there any plan to support TF2?

Or 

> The ARTPEC-8 DLPU uses per-tensor quantization, as opposed to per-axis quantization. This needs to be taken into account when choosing how to train the model, as the per-tensor quantization scheme strongly favours quantization-aware training. Because of this, quantization-aware training is used in the [training script](https://github.com/AxisCommunications/acap-native-sdk-examples/blob/main/tensorflow-to-larod-artpec8/env/training/train.py) for this tutorial. Currently, per-tensor quantization is not supported in Tensorflow 2.x, which is why the tf.contrib.quantize module from Tensorflow 1.15 is used to perform the quantization-aware training.

Does it mean TF2 is supported in axis if not using quantization ?

Corallo, 2023-09-14:
Hi @takuya-takeuchi 

Artpec-8 device support Tensorflow 2, but they are not optimized for the Tensorflow 2 quantization technique.
You could still try to use a model quantized with TF2, but you will encounter both speed and accuracy degradation. 
Depending on the type of model you use, this performance degradation can be more or less strong. 
So we do not recommend it.

You could however run a TF2 model with and without quantization using the CPU of the device, if you accept the lower speed.

pataxis, 2023-09-25:
Seems like the question was answered, will close the issue.

closed 2023-09-25 by pataxis