AxisCommunications/acap-native-sdk-examples #182: Unable to run app with high resolution

ajay-zessta, 2023-12-26:
getting this error on log

2023-12-25T22:48:08.644-07:00 axis-b8a44f742baa [ INFO    ] object_detection[551935]: Creating VDO High resolution image provider and stream 1280 x 720
2023-12-25T22:48:08.644-07:00 axis-b8a44f742baa [ INFO    ] object_detection[551935]: Dump of vdo stream settings map =====
2023-12-25T22:48:08.644-07:00 axis-b8a44f742baa [ INFO    ] object_detection[551935]: chooseStreamResolution: We select stream w/h=1280 x 720 based on VDO channel info.
2023-12-25T22:48:08.660-07:00 axis-b8a44f742baa [ ERR     ] object_detection[551935]: createStream: Failed creating vdo stream: DBus call failed: Failed creating stream: Unable to find a suitable buffer for 1280x720 NV12
2023-12-25T22:48:08.687-07:00 axis-b8a44f742baa [ ERR     ] object_detection[551935]: createImgProvider: Could not create VDO stream!
2023-12-25T22:48:08.710-07:00 axis-b8a44f742baa [ ERR     ] object_detection[551935]: main: Could not create high resolution image provider

shreyasatwork, 2023-12-29:
Hello,

Thanks for your question. Can you specify which camera model and firmware are you trying to run the application on?

ajay-zessta, 2024-01-02:
camera model: AXIS M2035-LE Bullet Camera
firmware: 10.12.199

shreyasatwork, 2024-01-02:
Thanks for sharing! Could you try the same with our latest firmware version 11.7.61? (You can find it here - https://www.axis.com/ftp/pub/axis/software/MPQT/M2035-LE/11_7_61/M2035-LE_11_7_61.bin)

Since there are most likely going to be breaking changes involved when there's a stepup in the major version of the firmware.

pataxis, 2024-02-12:
The examples use the latest SDK (1.12) which maps to a latest AXIS OS (11.8).

For older AXIS OS 10.12 versions you can find which SDK version that is compatible to a firmware in [compatibility](https://axiscommunications.github.io/acap-documentation/docs/axis-devices-and-compatibility/) page, in this case ACAP [4.3](https://axiscommunications.github.io/acap-documentation/docs/release-notes/4.3.html) release which has ACAP Native SDK tag 1.3.
As can also be seen in the release notes, this example repository had tag [1.3](https://github.com/AxisCommunications/acap-native-sdk-examples/tree/v1.3) for that release. Using the versioned SDK and example should hopefully work better.

closed 2024-03-21 by pataxis