AxisCommunications/acap-native-sdk-examples #183: object-detection example - quantization issue

Equidem, 2024-01-03:
While creating an object detection app based on the object-detection example here, I noticed that in the example, quantization of the input for the model isn't explicitly handled anywhere in the preprocessing setup, even though the model provided in the example has quantization on the input. Thus I thought that it had to be handled somewhere inside the Larod library, but from the testing I did with my own quantized model, this is apparently not the case, and I had to manually perform the quantization of the input before passing it to the model to make the model work.
I have two questions regarding this:
1. Is this a bug, or is the model provided with the example somehow special and doesn't require quantizing the input for it to work? Examining the model in Netron suggests otherwise, but maybe I am missing something.
2. For models that require the quantization, what would be the most optimal way to perform it using the C++ SDK libraries (VDO, Larod etc.)? I went through the available documentation, but couldn't find anything.

pataxis, 2024-01-12:
Hi @Equidem , thank you for your questions, we'll get back to you when we have had time to look in to it.

Equidem, 2024-01-15:
Hi @Corallo,
the issue here is not whether the input is in int8 (or more precisely uint8, since that's what VDO provides in the example.
As you can see in the attached screenshot from netron.app, the model used in the object-detection example in this repository ( https://github.com/google-coral/test_data/raw/master/ssd_mobilenet_v2_coco_quant_postprocess.tflite ) has quantization parameters of 128 as zero point and 0.0078125 as scale. From our testing, these are not applied to the input image anywhere in the example, since we had to manually use them to quantize the input image before passing it into our model to make it work. However, this seems like something that should be part of the preprocessing job, which is why I am asking whether there is some way to do it using the Larod or VDO libraries.
[image]

pataxis, 2024-02-08:
Closed due to inactivity

closed 2024-02-08 by pataxis