AxisCommunications/acap-native-sdk-examples #210: Invalid application manifest when adding dbus resource to manifest

Cacsjep, 2024-03-09:
Set dbus resource in manifest according to [docs](https://axiscommunications.github.io/acap-documentation/docs/develop/VAPIX-access-for-ACAP-applications.html), results in a non installable app.

```
2024-03-09T07:35:34.664+01:00 axis-b8a44f914a20 [ WARNING ] acapmanager[1579]: acapmanager-dbus.c:1770 Failed to install acap package '/tmp/pm_download_YR3EK2/package.eap' on location 'internal'
2024-03-09T07:35:34.664+01:00 axis-b8a44f914a20 [ WARNING ] acapmanager[1579]: acapmanager-dbus.c:629 acapctl error 80: Invalid application manifest
2024-03-09T07:35:34.664+01:00 axis-b8a44f914a20 [ WARNING ] packagemanager.cgi[1347185]: pm_install_acap.c:99 pm_install_acap: Acap manager D-Bus install method returned error: (1)

```
### To reproduce
Add resources and bus to manifest
```
{
    "schemaVersion": "1.3.1",
    "acapPackageConf": {
        "setup": {
            "appName": "dbustest",
            "friendlyName": "Dbus Test APP",
            "vendor": "Acs",
            "version": "1.0.1",
            "runMode": "never",
            "embeddedSdkVersion": "3.0",
            "user": {
                "group": "root",
                "username": "root"
            }
        }
    },
    "resources": {
        "dbus": {
            "requiredMethods": ["com.axis.HTTPConf1.VAPIXServiceAccounts1.GetCredentials"]
        }
    }
}
```

### Environment

- Axis device model: AXIS M3215-LVE
- Axis device firmware version: 11.9.60

Cacsjep, 2024-03-09:
oh I see I get the creds from dbus even i dont set the _resources dbus_ entry in the manifest.

pataxis, 2024-03-11:
Hi @Cacsjep , thanks for your question.

The documentation states that you need to use a dynamic user, the second bullet under updating the manifest in the [documentation](https://axiscommunications.github.io/acap-documentation/docs/develop/VAPIX-access-for-ACAP-applications.html#modify-the-manifest-file). This is done by removing the `user` section in the manifest.json.

Related to this in the ACAP 4.13 release last week,

- a new section explaining in more detail [dynamic user](https://axiscommunications.github.io/acap-documentation/docs/develop/application-project-structure.html#user-and-group) was added.
- a new example [vapix](https://github.com/AxisCommunications/acap-native-sdk-examples/tree/main/vapix) was added that shows VAPIX access from ACAP applications.

Cacsjep, 2024-03-12:
ok thx