AxisCommunications/acap-native-sdk-examples #221: Metadata Broker (aarch64) crash

paromarc, 2024-04-01:
### Bug report description

consume_analytics_metadata crashing on aarch64 (P3265-LV FW 11.9.60)

### To reproduce

I have been testing the Metadata Broker for few days and developed my own ACAP based on your example to find out my app was crashing after few hours of running. I have then decided to run the example app to find out it was crashing as well. The issue seems to be related somehow to the syslog. Commenting out 

syslog(LOG_INFO,
           "metadata received from topic: %s on source: %s: Monotonic time - "
           "%lld.%.9ld. Data - %.*s",
           channel_identifier->topic,
           channel_identifier->source,
           (long long)timestamp->tv_sec,
           timestamp->tv_nsec,
           (int)payload->size,
           (char*)payload->data);

results in the app never crashing. 

I have then decided to build a test acap flooding the camera with syslogs. The metadata broker example is the one that crashes while other acaps are not affected by it.

The same example 'consume_analytics_metadata' works just fine on armv7hf arch (tested on Q1615 MKIII fw 11.9.60)

The scene I am testing is a high traffic highway (see attached short video)

Steps to reproduce the behavior:

1. Install the 'consume_analytics_metadata_1_0_0' acap on ARTPEC8 camera (I tested on P3265-LV) FW 11.9.60
2. Make sure the camera views faces lots of traffic (the more, the quicker the issue will happen)
3. Check after few hours of running that the app stopped.

### Screenshots

Check video.
https://github.com/AxisCommunications/acap-native-sdk-examples/assets/11927582/919b3739-a577-4c3b-8a94-6ff552b4ca13

### Environment

- Axis device model: P3265-LV
- Axis device firmware version: 11.9.60
- Stack trace or logs: Server Report attached
[p3265_metadata_broker_issue.txt](https://github.com/AxisCommunications/acap-native-sdk-examples/files/14825509/p3265_metadata_broker_issue.txt)

- OS and version: na
- Version: example compiled iusing acap-native-sdk v1.13

### Additional context

As mentioned above the issue can be avoided by commenting out the syslog, so this issue doesn't have a big impact on my current dev. but definitely worth investigating in my opinion.

pataxis, 2024-04-02:
Thanks @paromarc  for your report, I will forward it.

kevin-barudde-work, 2024-08-19:
Hi @paromarc,

Sorry for the wait, but  we are not able to recreate your bug.
We have run the ACAP on a P3265-V with version 11.9.60, looking at a busy scene in over a week without it failing.

Can you still recreate the bug?
If yes, have you made any modifications to the ACAP or had something else running on the camera?

paromarc, 2024-08-20:
Hey Kevin, thanks for testing on your end ! I defaulted my P3265-LV and I cannot duplicate on my end either. Maybe something weird happening on my camera. I mess with it a lot, so I am not surprised :-) I think we can close this for now :-)
Thanks again for checking on your end !

closed 2024-08-22 by pataxis