AxisCommunications/acap-native-sdk-examples #222: Errors in logs with axoverlay example

Equidem, 2024-04-01:
When I build and run the axoverlay example on our camera (no changes to the code made) it produces two errors in the app log:
[ WARNING ] axoverlay[2777417]: g_error_free: assertion 'error != NULL' failed
[ INFO    ] axoverlay[2777417]: Fontconfig error: No writable cache directories

The errors don't seem to interfere with the function of the app which is running as it should, including drawing the correct overlays, but I assume the app shouldn't be producing any errors at all, and there may be consequences I am not thinking of to the errors, which is why I am reporting them.

pataxis, 2024-04-08:
Hi @Equidem , thanks for reporting this, we'll get back when we have had time to test it.

closed 2024-04-09 by pataxis

Cacsjep, 2025-01-25:
also get the Fontconfig error: No writable cache directories, whats the reason for that ?

commit c73e38c4 (2024-04-09): Fix/warning logs in axoverlay example

Fixes #222
commit f05207bf (2024-04-09): Fix/warning logs in axoverlay example

Fixes #222