AxisCommunications/acap-native-sdk-examples #232: cpu-proc padding

Cacsjep, 2024-04-25:
### Larod cpu-proc padding

It would be cool when padding trough larod preproccessing is possible instead of croping a image.

Like when getting an vdo frame of 640*360 to padding this Image with black Pixels to 640*640 for yolo for example.

Maybe its already implemented but i dont found any Infos in docs about this.

Steps would be:
   - create a larod map like for cropping but for padding.

pataxis, 2024-08-23:
Hi @Cacsjep , sorry for the late response, thanks for the suggestion, we'll have it mind for future maintenance and updates to the example.

isak-jakobsson, 2025-02-13:
Hi @Cacsjep, from AXIS OS 12.2 and ACAP SDK 12.2, it is possible to do this type of cropping.

But, instead of using the pro-processing through larod, it has been implemented directly to vdo. 

To request a padded square stream, set the new key value ("square") on the VdoMap before creating the VdoStream:
`vdo_map_set_uint32(vdoMap, "square", 640);`
This will produce a stream of the desired resolution, e.g., 640x640 in this case.

closed 2025-02-13 by Cacsjep