AxisCommunications/acap-native-sdk-examples #264: Model load error on CV25

AtsushiYokokawa, 2024-07-01:
### Discussed in https://github.com/AxisCommunications/acap-native-sdk-examples/discussions/263

<div type='discussions-op-text'>

<sup>Originally posted by **AtsushiYokokawa** June 26, 2024</sup>
The model converted with tensorflow-to-larod-cv25 does not work.
Here are the steps I tried.

- Environment
    - PC
        - OS: Windows 11 Pro (WSL: Ubuntu-20.04)
        - CPU: 11th Gen Intel(R) Core(TM) i7-11390H @ 3.40GHz   2.92 GHz
        - RAM: 16.0GB
        - GPU: Intel IRIS
        - Docker: Docker Desktop V4.17.0
        - CVTool: Ambarella_Toolchain_CNNGen_Basic_2.5.6_20231117
    - Camera
        - Camera: M3086-V
        - FW: 11.10.83

- Steps
    - Model conversion
        ```
        docker build --tag tensorflow-to-larod-cv25 --build-arg TRAIN_EPOCHS=1 .
        docker cp $(docker create tensorflow-to-larod-cv25):/env/training/models/converted_model.tflite ./converted_model.tflite
        ```
    - Compile with CVTool
        ```
        gen_image_list.py -f pics -o dra_image_bin/img_list.txt -ns -e .jpg -c 0 -d 0,0 -r 256,256 -bf dra_image_bin -bo dra_image_bin/dra_bin_list.txt
        # [gen_image_list] Folder input received: pics
        # [gen_image_list] Created file: dra_image_bin/img_list.txt
        # (info) using #30 thread(s) to run imgtobin
        # [info][#13] Convert 12a768fce3b51894.jpg to 12a768fce3b51894.bin ... Done
        # [info][ #0] Convert 00ee5d5169b5d94f.jpg to 00ee5d5169b5d94f.bin ... Done
        # [info][ #5] Convert 027402a2f2de4ffd.jpg to 027402a2f2de4ffd.bin ... Done
        # [info][ #1] Convert 016782dc5cdbd0ce.jpg to 016782dc5cdbd0ce.bin ... Done
        # [info][ #7] Convert 042926ab7d9c4922.jpg to 042926ab7d9c4922.bin ... Done
        # [info][ #2] Convert 01bce5b2dd400f5d.jpg to 01bce5b2dd400f5d.bin ... Done
        # [info][ #8] Convert 07afa437e2aa458b.jpg to 07afa437e2aa458b.bin ... Done
        # [info][#10] Convert 09f03d43f0a2a3aa.jpg to 09f03d43f0a2a3aa.bin ... Done
        # [info][ #9] Convert 07d343487bd82d5a.jpg to 07d343487bd82d5a.bin ... Done
        # [info][#11] Convert 1186a1ccc9923678.jpg to 1186a1ccc9923678.bin ... Done
        # [info][ #6] Convert 031ce4a78c6b7331.jpg to 031ce4a78c6b7331.bin ... Done
        # [info][#14] Convert 12ad6bec64eb4454.jpg to 12ad6bec64eb4454.bin ... Done
        # [info][ #3] Convert 01f969a887724083.jpg to 01f969a887724083.bin ... Done
        # [info][ #4] Convert 020c12da1d0ea78c.jpg to 020c12da1d0ea78c.bin ... Done
        # [info][#17] Convert 15f9857bc6e82d05.jpg to 15f9857bc6e82d05.bin ... Done
        # [info][#12] Convert 11ee142c27bf2f3e.jpg to 11ee142c27bf2f3e.bin ... Done
        # [info][#15] Convert 13997536ad9d92ef.jpg to 13997536ad9d92ef.bin ... Done
        # [info][ #2] Convert 22bc484961df0de7.jpg to 22bc484961df0de7.bin ... Done
        # [info][ #9] Convert 26742912e6591e49.jpg to 26742912e6591e49.bin ... Done
        # [info][#10] Convert 24b4f02a4c49cdca.jpg to 24b4f02a4c49cdca.bin ... Done
        # [info][#20] Convert 2f9a9dac02715200.jpg to 2f9a9dac02715200.bin ... Done
        # [info][#11] Convert 28bc743ce467c9b2.jpg to 28bc743ce467c9b2.bin ... Done
        # [info][#13] Convert 164e8d631323dda2.jpg to 164e8d631323dda2.bin ... Done
        # [info][#17] Convert 345d112cdb3d621c.jpg to 345d112cdb3d621c.bin ... Done
        # [info][ #9] Convert 43466129aba65f94.jpg to 43466129aba65f94.bin ... Done
        # [info][#12] Convert 3627ae4bceb27c2b.jpg to 3627ae4bceb27c2b.bin ... Done
        # [info][#16] Convert 14bb7d41e1cd3b27.jpg to 14bb7d41e1cd3b27.bin ... Done
        # [info][#14] Convert 2e3eb2b6dbae8ac5.jpg to 2e3eb2b6dbae8ac5.bin ... Done
        # [info][ #9] Convert 62642779b29a0b0e.jpg to 62642779b29a0b0e.bin ... Done
        # [info][ #7] Convert 22bb3097d0c0b214.jpg to 22bb3097d0c0b214.bin ... Done
        # [info][#19] Convert 25d6711130edac64.jpg to 25d6711130edac64.bin ... Done
        # [info][#21] Convert 39652e445524e644.jpg to 39652e445524e644.bin ... Done
        # [info][ #5] Convert 1c9c30bbe43ff9b0.jpg to 1c9c30bbe43ff9b0.bin ... Done
        # [info][#20] Convert 4c746ea68c15eca5.jpg to 4c746ea68c15eca5.bin ... Done
        # [info][ #0] Convert 1a1e1cadeb581a60.jpg to 1a1e1cadeb581a60.bin ... Done
        # [info][#15] Convert 3dbf4fea124969b1.jpg to 3dbf4fea124969b1.bin ... Done
        # [info][#18] Convert 17758ba0e16314b0.jpg to 17758ba0e16314b0.bin ... Done
        # [info][#12] Convert 63e86e3144157862.jpg to 63e86e3144157862.bin ... Done
[... 143 lines cut ...]
        # car_human_model: DONE analyzing
        # car_human_model: DONE opt
        # car_human_model: DONE buffering
        # car_human_model: SPLIT 0 (split progress: 46%)
        # car_human_model: SPLIT 1 (split progress: 55%)
        # car_human_model: SPLIT 2 (split progress: 100%)
        # car_human_model: DONE split
        # car_human_model: DONE all
        ```
        ```
        cavalry_gen -d vas_output -f car_human_model_cavalry.bin
        # Using cavalry_gen version latest
        # 
        # Save to file car_human_model_cavalry.bin with file size 3319784.
        ```
    - Test model on CV25 camera
        ```
        larod-client -g ./car_human_model_cavalry2.bin  -c ambarella-cvflow -R 10 -w 5 -i ''
        # 2024-06-26T17:54:07.971 Connecting to larod...
        # 2024-06-26T17:54:07.981 Connected
        # 2024-06-26T17:54:08.030 ERROR: When loading model synchronously (-2): Could not load model: Could not initialize net
        ```

- Resources
    - https://mega.nz/file/Fnl2mAab#5Pq8zQRyhuKy2X8rWPPSTZt43ma3PpAzleoRDaAB0h0</div>

AtsushiYokokawa, 2024-07-01:
As stated in the discussion, it seems that compatibility between AXIS OS and the Ambarella Toolchain has not been ensured.

Please check.

pataxis, 2024-07-16:
Hi @AtsushiYokokawa , thank you for your feedback. We'll get back when we had time to investigate it.