AxisCommunications/acap-native-sdk-examples #271: error while loading shared libraries: libmdb.so.0

amotivv, 2024-07-08:
### Describe the bug

Running example metadata-broker/consume-analytics-scene-description application on AXIS D2110-VE Security Radar produces the following error in the app logs on the device:

```
[ INFO    ] consume_analytics_metadata[25839]: /usr/local/packages/consume_analytics_metadata/consume_analytics_metadata: error while loading shared libraries: libmdb.so.0: cannot open shared object file: No such file or directory
```

### To reproduce

Build Docker image using example Dockerfile in current repo:

```docker build --tag consume_analytics_metadata:1.0 --build-arg ARCH=armv7hf .```

Copy application:

```docker cp $(docker create consume_analytics_metadata:1.0):/opt/app ./build```

Install ```consume_analytics_metadata_1_0_0_armv7hf.eap``` application to  D2110-VE.
Start application.
View App logs:
```
----- Contents of SYSTEM_LOG for 'consume_analytics_metadata' -----

2024-07-08T12:30:46.456-04:00 axis-b8a44f809e92 [ INFO    ] consume_analytics_metadata[22581]: /usr/local/packages/consume_analytics_metadata/consume_analytics_metadata: error while loading shared libraries: libmdb.so.0: cannot open shared object file: No such file or directory
2024-07-08T12:34:29.712-04:00 axis-b8a44f809e92 [ INFO    ] consume_analytics_metadata[23520]: /usr/local/packages/consume_analytics_metadata/consume_analytics_metadata: error while loading shared libraries: libmdb.so.0: cannot open shared object file: No such file or directory
2024-07-08T12:56:33.467-04:00 axis-b8a44f809e92 [ INFO    ] consume_analytics_metadata[25839]: /usr/local/packages/consume_analytics_metadata/consume_analytics_metadata: error while loading shared libraries: libmdb.so.0: cannot open shared object file: No such file or directory
2024-07-08T13:08:42.712-04:00 axis-b8a44f809e92 [ INFO    ] consume_analytics_metadata[27049]: /usr/local/packages/consume_analytics_metadata/consume_analytics_metadata: error while loading shared libraries: libmdb.so.0: cannot open shared object file: No such file or directory
2024-07-08T13:17:14.962-04:00 axis-b8a44f809e92 [ INFO    ] consume_analytics_metadata[28293]: /usr/local/packages/consume_analytics_metadata/consume_analytics_metadata: error while loading shared libraries: libmdb.so.0: cannot open shared object file: No such file or directory
```

### Screenshots

[image]

### Environment

- Axis device model: D2110-VE
- Axis device firmware version: 11.10.61
- Stack trace or logs: Shown above
- OS and version: macOS 14.5 (23F79)

### Additional context

Building and installing the ```hello-world``` sample app works without issue when installed on the D2110-VE.

Cacsjep, 2024-07-08:
I dont know excalty about mdb Support for this device but when the .so is not there then normally its not supported by the device

amotivv, 2024-07-08:
Thank you. Built the same application for a M3065-V (aarch64) with 10.12.240 firmware and got the same:
```
2024-07-08T15:32:49.395-04:00 axis-b8a44f078bac [ INFO    ] consume_analytics_metadata[3208]: /usr/local/packages/consume_analytics_metadata/consume_analytics_metadata: error while loading shared libraries: libmdb.so.0: cannot open shared object file: No such file or directory
```
Is this an issue with the [acap-native-sdk](https://hub.docker.com/r/axisecp/acap-native-sdk), or have I just picked the wrong devices to test with?

Cacsjep, 2024-07-08:
Did you read about the API Firmware Version where its added?  10.12.240 is to low. https://axiscommunications.github.io/acap-documentation/docs/api/native-sdk-api.html#metadata-broker-api

amotivv, 2024-07-08:
> Did you read about the API Firmware Version where Mdb is added to Axis OS? 10.12.240 is to low. https://axiscommunications.github.io/acap-documentation/docs/api/native-sdk-api.html#metadata-broker-api

Yes, for the camera. Was just trying to reproduce the issue. I bet the mdb is not available in the radar. Thanks for the help.

amotivv, 2024-07-08:
Good tip. Looks it's not there in the radar:
```
root@axis-b8a44f809e92:/# find . -type f | grep libmdb
root@axis-b8a44f809e92:/#
```

pataxis, 2024-07-09:
Hi @amotivv , thanks for your detailed report.

The SoC is ARTPEC-7 which is marked as supported for the Metadata Broker API.
I think this Beta API has so far been focusing on information on detected objects on a video stream from an image sensor and not a radar and could be a reason that it's not included on the product.

But it's a good question and I will look in to it.

apljungquist, 2024-07-09:
If I remember correctly the broker is so far present only on products where one of the [standard streams](https://axiscommunications.github.io/acap-documentation/docs/api/src/api/metadata-broker/html/standard_topics.html) are produced. These currently coincide with the presence of the metadata producer labeled _Analytics Scene Description_; see `/camera/index.html#/system/metadataConfig` on your device. Unfortunately, I don't have a D2110-VE at hand to verify, so if it seems I may be wrong then I probably am.

pataxis, 2024-08-23:
Hi again @amotivv , the pure radar product D2110 doesn't have support for Metadata Broker API today because the current topics 

- `com.axis.analytics_scene_description.v0.beta`
- `com.axis.consolidated_track.v1.beta`

are only related to detections from a video stream.

The [Scene Metadata](https://www.axis.com/products/axis-scene-metadata#compatible-products) page on axis.com has a list of products that are supported for the current topics. There are two Radar-Video Fusion Cameras on the list.

closed 2024-08-23 by pataxis