AxisCommunications/acap-native-sdk-examples #272: Error libm.so.6: version `GLIBC_2.38' & libstdc++.so.6: version `GLIBCXX_3.4.30' not found

Wiphat101, 2024-07-09:
I tested the application using_opencv from the example https://github.com/AxisCommunications/acap-native-sdk-examples/tree/main/using-opencv
I haven't made any changes yet. When I test the .eap file with the camera, I get an error in the app log.
**Commands I use to build application**
docker build --tag <APP_IMAGE> .

Error #1
/usr/local/packages/opencv_app/opencv_app: /usr/lib/libm.so.6: version `GLIBC_2.38' not found (required by /usr/local/packages/opencv_app/lib/libopencv_imgproc.so.4.5)

Error #2
/usr/local/packages/opencv_app/opencv_app: /usr/lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/local/packages/opencv_app/lib/libopencv_core.so.4.5)

[image]

What causes this problem and what do I need to fix?

Cacsjep, 2024-07-09:
Which camera Version and SDK Version you are using ?

vivekatoffice, 2024-07-09:
I agree with @Cacsjep. You can find the [ACAP Native SDK software compatibility](https://axiscommunications.github.io/acap-documentation/docs/axis-devices-and-compatibility/#acap-native-sdk-software-compatibility) list

[image]

pataxis, 2024-07-09:
Hi @Wiphat101 , as both @Cacsjep  and @vivekatoffice  have hinted, this error is a compatibility issue.

This repository always show-case the use of the latest released ACAP Native SDK.
In the [compatibility](https://axiscommunications.github.io/acap-documentation/docs/axis-devices-and-compatibility/#acap-native-sdk-software-compatibility) table you can find which SDK to use for a certain AXIS OS version.

If you want to have example applications that work for a certain SDK release, go to the [Release](https://github.com/AxisCommunications/acap-native-sdk-examples/releases) page of this repo and click the tag to browse the repository at that point of time.

A more in-depth description of the compatibility issue can be found [here](https://github.com/AxisCommunications/acap-native-sdk-examples/discussions/59) before.

closed 2024-07-15 by pataxis