AxisCommunications/acap-native-sdk-examples #276: Examples do not work as intended.

CoolBassist, 2024-08-19:
### Describe the bug

I have noticed that trying to execute some of the examples from the "acap-native-sdk-examples" repo require some extra editing to get working. For example, download the files from the "axserial" example, and then follow the instructions exactly. When I did this, it just simply copied the files from `./app` to `./build`. To fix this I needed to edit the last line of the docker file to this :-
 ```RUN . /opt/axis/acapsdk/environment-setup* && acap-build .```

Another example is the "axStorage" example. When running it, I had the following error in the appLog.
```
Failed to list storage devices. Error: (GDBus.Error:org.freedesktop.DBus.Error.AccessDenied: Rejected send message, 3 matched rules; type="method_call", sender=":1.407" (uid=997 pid=1960 comm="/usr/local/packages/axstorage/axstorage") interface="com.axis.Storage" member="GetAllStoragesInfo" error name="(unset)" requested_reply="0" destination=":1.205" (uid=211 pid=1257 comm="/usr/bin/storage_manager -n --serial-no B8A44F76D8"))
```
To solve this, I had to remove the "resources" object from the manifest, and instead use the "user" object with values of "sdk".
 
### To reproduce
Download all files from a repo such as "axSerial", then follow the instructions exactly.

### Environment
WSL: Ubuntu 22.04
Docker version 24.0.7, build 24.0.7-0ubuntu2~22.04.1

apljungquist, 2024-08-19:
What firmware version are you targetting?

I experienced something similar with the axevent API and if I remember correctly I concluded that on versions prior to around 11.5 one has to use the sdk user, and on later firmwares one has use dynamic users. Not sure if [this](https://axiscommunications.github.io/acap-documentation/docs/develop/user-selection.html) is the best page for explaining it, but it does talk about the difference.

pataxis, 2024-08-23:
Hi @CoolBassist , thanks for reaching out!

> I have noticed that trying to execute some of the examples from the "acap-native-sdk-examples" repo require some extra editing to get working. 

When you say executing, do you mean building the Docker container image with the following command?
`docker build --tag <APP_IMAGE> .`

If you mean that you had to replace the four last lines in axstorage [Dockerfile](https://github.com/AxisCommunications/acap-native-sdk-examples/blob/8793748fe894c835f7ee42912bdfb7b866428d3e/axstorage/Dockerfile#L14-L17), that is the [here-documents](https://docs.docker.com/reference/dockerfile/#here-documents) (or heredocs) syntax (set on the first line with `# syntax=docker/dockerfile:1`) and it sounds like it could be that your Docker environment doesn't support it by default.

I do however think that Docker 24.04 should have support for it. You can test to set this environment variable in connection to build.
```sh
DOCKER_BUILDKIT=1 docker build --tag <APP_IMAGE> .
```
and/or you can build with buildx
```sh
docker buildx build --tag <APP_IMAGE> .
```

closed 2024-09-17 by pataxis