AxisCommunications/acap-native-sdk-examples #293: Trying to build the web-server example, has issues with docker

lbRaytec, 2024-09-20:
Hello, I am trying to work with the web-server example. I pulled the code from GitHub repo and executed the following Docker command to build the example.

`docker build . --build-arg ARCH=armv7hf --tag web-server:armv7hf`

The docker execution resulted in the following error.

`=> ERROR [ 6/10] RUN git apply --ignore-space-change --ignore-whitespace ../monkey.patch &&    . /opt/axis/acapsdk/envi  0.2s
------
 > [ 6/10] RUN git apply --ignore-space-change --ignore-whitespace ../monkey.patch &&    . /opt/axis/acapsdk/environment-setup* &&    ./configure     --enable-shared     --malloc-libc     --prefix=/usr/local     --bindir=/usr/local/bin     --libdir=/usr/local/lib     --sysconfdir=/usr/local/packages/list/html     --datadir=/usr/local/packages/list/html     --mandir=/usr/local/man     --logdir=/tmp     --plugdir=/usr/local/packages/list/lib     --pidfile=/tmp/monkey.pid     --incdir=/usr/local/include/monkey     --systemddir=/usr/lib/systemd/system &&    make &&    make install:
0.217 error: corrupt patch at line 6
------
Dockerfile:15
--------------------
  14 |     WORKDIR /opt/monkey
  15 | >>> RUN git apply --ignore-space-change --ignore-whitespace ../monkey.patch &&\
  16 | >>>     . /opt/axis/acapsdk/environment-setup* &&\
  17 | >>>     ./configure \
  18 | >>>     --enable-shared \
  19 | >>>     --malloc-libc \
  20 | >>>     --prefix=/usr/local \
  21 | >>>     --bindir=/usr/local/bin \
  22 | >>>     --libdir=/usr/local/lib \
  23 | >>>     --sysconfdir=/usr/local/packages/list/html \
  24 | >>>     --datadir=/usr/local/packages/list/html \
  25 | >>>     --mandir=/usr/local/man \
  26 | >>>     --logdir=/tmp \
  27 | >>>     --plugdir=/usr/local/packages/list/lib \
  28 | >>>     --pidfile=/tmp/monkey.pid \
  29 | >>>     --incdir=/usr/local/include/monkey \
  30 | >>>     --systemddir=/usr/lib/systemd/system &&\
  31 | >>>     make &&\
  32 | >>>     make install
  33 |
--------------------
ERROR: failed to solve: process "/bin/sh -c git apply --ignore-space-change --ignore-whitespace ../monkey.patch &&    . /opt/axis/acapsdk/environment-setup* &&    ./configure     --enable-shared     --malloc-libc     --prefix=/usr/local     --bindir=/usr/local/bin     --libdir=/usr/local/lib     --sysconfdir=/usr/local/packages/list/html     --datadir=/usr/local/packages/list/html     --mandir=/usr/local/man     --logdir=/tmp     --plugdir=/usr/local/packages/list/lib     --pidfile=/tmp/monkey.pid     --incdir=/usr/local/include/monkey     --systemddir=/usr/lib/systemd/system &&    make &&    make install" did not complete successfully: exit code: 128`

Am I wrong with versions here because that's the only possible scenario I could think off. As, I have just pulled the code from example and trying to execute it.

TIA!

pataxis, 2024-09-27:
Hi @lbRaytec , thanks for your question.

The example builds and test pass in nightly tests, so more information about your system would be required to see if the problem can be replicated.

What OS and OS version are you using?
Which version of Docker Desktop or Docker Engine are you using?
Run `docker --version` and paste command and result here

Are you standing on `main` branch of this repository when building? Could you paste the SHA1 of last commit or run `git log -1` and paste the command and output here?
  
Have you modified the web-server example?

lbRaytec, 2024-09-27:
Windows 11 version 23H2
Docker version (Docker version 27.2.0, build 3ab4256)

commit bea36cadd81ca7a14ae0df52920018a5b94736a6 (HEAD -> main, origin/main, origin/HEAD)
Author: renovate[bot] <29139614+renovate[bot]@users.noreply.github.com>
Date:   Tue Sep 17 15:40:17 2024 +0000

    chore(deps): update super-linter/super-linter action to v7
    
    
    No, I have not modified the example. did `git clone <url>`.  Then, navigated into the `web-server` example and ran as per the documentation.

pataxis, 2024-09-27:
Are you using Docker Desktop for Windows and building there or are you using Windows Subsystem for Linux (WSL) with Ubuntu?

lbRaytec, 2024-09-27:
Navigated to `web-server` folder.

     `set ARCH=armv7hf`

Have to use `set` here because it is the windows alternate for `export`.

     docker build . --build-arg ARCH --tag web-server:ARCH
     
 This gives me the error!

lbRaytec, 2024-09-30:
Hi, I tested with WSL and it still reproduces the same error. 

`docker build . --build-arg ARCH --tag web-server:$ARCH
[+] Building 7.8s (11/15)                                                                                                                                                                         docker:default
 => [internal] load build definition from Dockerfile                                                                                                                                                        0.1s
 => => transferring dockerfile: 1.51kB                                                                                                                                                                      0.0s
 => [internal] load metadata for docker.io/axisecp/acap-native-sdk:1.15-armv7hf-ubuntu22.04                                                                                                                 2.9s
 => [auth] axisecp/acap-native-sdk:pull token for registry-1.docker.io                                                                                                                                      0.0s
 => [internal] load .dockerignore                                                                                                                                                                           0.1s
 => => transferring context: 2B                                                                                                                                                                             0.0s
 => [ 1/10] FROM docker.io/axisecp/acap-native-sdk:1.15-armv7hf-ubuntu22.04@sha256:0ba1915b2f6b175e81ff2229a1f9057509e2541a92110856e98de10b376f179a                                                         0.0s
 => [internal] load build context                                                                                                                                                                           0.1s
 => => transferring context: 30.20kB                                                                                                                                                                        0.1s
 => CACHED [ 2/10] WORKDIR /opt                                                                                                                                                                             0.0s
 => [ 3/10] COPY monkey.patch .                                                                                                                                                                             0.1s
 => [ 4/10] RUN git clone -b v1.5.6 https://github.com/monkey/monkey                                                                                                                                        4.0s
 => [ 5/10] WORKDIR /opt/monkey                                                                                                                                                                             0.1s
 => ERROR [ 6/10] RUN git apply --ignore-space-change --ignore-whitespace ../monkey.patch &&    . /opt/axis/acapsdk/environment-setup* &&    ./configure     --enable-shared     --malloc-libc     --prefi  0.4s
------
 > [ 6/10] RUN git apply --ignore-space-change --ignore-whitespace ../monkey.patch &&    . /opt/axis/acapsdk/environment-setup* &&    ./configure     --enable-shared     --malloc-libc     --prefix=/usr/local     --bindir=/usr/local/bin     --libdir=/usr/local/lib     --sysconfdir=/usr/local/packages/list/html     --datadir=/usr/local/packages/list/html     --mandir=/usr/local/man     --logdir=/tmp     --plugdir=/usr/local/packages/list/lib     --pidfile=/tmp/monkey.pid     --incdir=/usr/local/include/monkey     --systemddir=/usr/lib/systemd/system &&    make &&    make install:
0.328 error: corrupt patch at line 6
------
Dockerfile:15
--------------------
  14 |     WORKDIR /opt/monkey
  15 | >>> RUN git apply --ignore-space-change --ignore-whitespace ../monkey.patch &&\
  16 | >>>     . /opt/axis/acapsdk/environment-setup* &&\
  17 | >>>     ./configure \
  18 | >>>     --enable-shared \
  19 | >>>     --malloc-libc \
  20 | >>>     --prefix=/usr/local \
  21 | >>>     --bindir=/usr/local/bin \
  22 | >>>     --libdir=/usr/local/lib \
  23 | >>>     --sysconfdir=/usr/local/packages/list/html \
  24 | >>>     --datadir=/usr/local/packages/list/html \
  25 | >>>     --mandir=/usr/local/man \
  26 | >>>     --logdir=/tmp \
  27 | >>>     --plugdir=/usr/local/packages/list/lib \
  28 | >>>     --pidfile=/tmp/monkey.pid \
  29 | >>>     --incdir=/usr/local/include/monkey \
  30 | >>>     --systemddir=/usr/lib/systemd/system &&\
  31 | >>>     make &&\
  32 | >>>     make install
  33 |
--------------------
ERROR: failed to solve: process "/bin/sh -c git apply --ignore-space-change --ignore-whitespace ../monkey.patch &&    . /opt/axis/acapsdk/environment-setup* &&    ./configure     --enable-shared     --malloc-libc     --prefix=/usr/local     --bindir=/usr/local/bin     --libdir=/usr/local/lib     --sysconfdir=/usr/local/packages/list/html     --datadir=/usr/local/packages/list/html     --mandir=/usr/local/man     --logdir=/tmp     --plugdir=/usr/local/packages/list/lib     --pidfile=/tmp/monkey.pid     --incdir=/usr/local/include/monkey     --systemddir=/usr/lib/systemd/system &&    make &&    make install" did not complete successfully: exit code: 128`

lbRaytec, 2024-10-03:
Hi, just a quick update on this one. I created an Ubuntu 22.04 LTS set up in VM. I then pulled the code and ran the web-server example there and it worked. I am wondering there is something wrong when it comes to Windows may be? As you previously mentioned that it works for Mac OS. 

I tried with three different Windows system and all resulted in the same error. 

All the three Windows system has same OS version. docker version.

However, I will further generate EAP file and see how it goes! will keep you posted. Thanks for your time!

pataxis, 2024-10-03:
> As you previously mentioned that it works for Mac OS.
We have tests on Linux machines which works. I'm not sure if Docker Desktop on Mac OS has the same issue.

closed 2025-02-18 by GHakansson