AxisCommunications/acap-native-sdk-examples #300: how to disable the syslog of boundingbox?

rocky-ji, 2024-10-08:
When trying the boundingbox sample on P3265 and Q3538 OS 11.11.73, the bounding box really works in camera web GUI, but it seems in the camera server reports, it will keep output following log, is there any way to close this log?

2024-10-08T16:32:28.722+08:00 axis-b8a44f90d12b [ INFO    ] overlayd[1096]: /usr/local/packages/bounding_box_example/bounding_box_example vanished, removing its overlays
2024-10-08T16:32:28.723+08:00 axis-b8a44f90d12b [ INFO    ] overlayd[1096]: :1.880 vanished, tearing down command channel
2024-10-08T16:32:28.760+08:00 axis-b8a44f90d12b [ INFO    ] overlayd[1096]: Setting up command channel for :1.881
2024-10-08T16:32:33.797+08:00 axis-b8a44f90d12b [ INFO    ] overlayd[1096]: /usr/local/packages/bounding_box_example/bounding_box_example vanished, removing its overlays
2024-10-08T16:32:33.798+08:00 axis-b8a44f90d12b [ INFO    ] overlayd[1096]: :1.881 vanished, tearing down command channel
2024-10-08T16:32:33.841+08:00 axis-b8a44f90d12b [ INFO    ] overlayd[1096]: Setting up command channel for :1.882
2024-10-08T16:32:38.881+08:00 axis-b8a44f90d12b [ INFO    ] overlayd[1096]: :1.882 vanished, tearing down command channel
2024-10-08T16:32:38.882+08:00 axis-b8a44f90d12b [ INFO    ] overlayd[1096]: /usr/local/packages/bounding_box_example/bounding_box_example vanished, removing its overlays

pataxis, 2024-10-11:
Hi @rocky-ji , I'm not sure I have seen this. 

@daniel-nielsen-se have you seen this?

rocky-ji, 2024-10-11:
I tried on P3268 and Q3538, if you open the video on camera web GUI and see the boxing box, then in server reports, there is a lot  logs as i listed above.

daniel-nielsen-se, 2024-10-11:
I can reproduce. However it's not a bug in bbox itself. axoverlay produces the same logs.

It's a firmware debug feature, that probably should have used DEBUG log level instead of INFO. We can however forward the feedback to the team responsible for **overlayd**. 

However please note that during normal operation, the intention is that you start your bbox application once per boot, then draw on demand, so even though the prints are verbose they should not flood the log continuously.

closed 2024-10-11 by pataxis