AxisCommunications/acap-native-sdk-examples #313: Can't make docker image of tensorflow-to-larod-artpec8

Ajiro3, 2024-12-19:
> **Please do not disclose security vulnerabilities as is
[image]
sues. See our [security policy](../../SECURITY.md) for responsible disclosures.**

### Describe the bug
I tried to buld docker image of "tensorflow-to-larod-artpec8" but failed.
I invoke the docker desktop to build the image as suggested in "How to run the code" in README.
  docker build --tag tfmodel_train_a8 . 
But it showed the following error.

Explain the behavior you would expect and the actual behavior.
## My purpose 
I want to build the docker image of object detection inferencewith  python scripts to test its capability on a P1467 Cam.
Building the docker image of "hello-world" and loading it on P1467 Cam is easy.
But there is no sample nor documents how to build a docker image of python scripts with acap for P1467 Cam.
Trying to write a Dockerfile for building  .eap file is terrible work.

### To reproduce

Please provide as much context as possible and describe the *reproduction steps* that someone else can follow to recreate the issue on their own.

A team member will try to reproduce the issue with your provided steps. If there are no reproduction steps or no obvious way to reproduce the issue, the team will ask you for those steps. Bugs without steps will not be addressed until they can be reproduced.

Steps to reproduce the behavior:

1. Set up '...'
2. Do this '...'
3. See error

### Screenshots

If applicable, add screenshots to help explain your problem.

### Environment

- Axis device model: P1467
- Axis device firmware version: [11.3.70]
- Stack trace or logs: [e.g. Axis device system logs]
- OS and version: [ Ubuntu 20.04 LTS, Windows 10 Pro 22H2]
- Version: [version of the application, SDK, runtime environment, package manager, interpreter, compiler, depending on what seems relevant]

### Additional context

Add any other context about the problem here.

Cacsjep, 2024-12-19:
Ah did you not see that for pyrhon you should use sdk4 and not native sdk

Ajiro3, 2024-12-21:
Thank you for your response.
I should work on https://github.com/AxisCommunications/acap-computer-vision-sdk-examples, correct?

WilfR, 2024-12-21:
Hi Ajiro, 
  It seems that I can build the latest version of tensorflow-to-larod-artpec8 from the acap-native-sdk-examples directory on my Windows computer just fine. Is it possible that you are trying to build an old version of the project? If so please get the latest version of the acap-native-sdk-examples repository and try again. (Beware that it will take a long time to build this specific project because as part of the build process the software will have to train an AI model from scratch. The build output will keep you updated on the progress, but it might take an hour or more to execute depending on your computer.) 
Cheers, 
Wilf.

Cacsjep, 2024-12-21:
@WilfR i think he would use python not C

WilfR, 2024-12-21:
Hi @Cacsjep, 
  I am not sure that I understand your comment. As I understand @Ajiro3's objective, he is trying to build the Docker image from the tensorflow-to-larod-artpec8 example in the acap-native-sdk-examples repository. The purpose of this example is to show how to create and train a .tflite model within a Docker container. Within the Docker image that is being built, python is being used to run the training and convert_model scripts. That all seems fine to me - am I missing something? 
  Please note that, just so we are on the same page, this particular example is NOT trying to create an ACAP application, it is just creating a model ("converted_model.tflite") that may be used in some other ACAP application. There is nothing that has anything to do with any AXIS SDK here. 
  Or have I misunderstood your reply? 
Cheers, 
Wilf.

Ajiro3, 2024-12-21:
Hi @WilfR-san and  @Cacsjep-san,

Thank you for your response.
License Plate Recognition tool by AXIS is not applicable for two linse type plate in Japan.
I developed a Yolo V3 tiny model to detect Japanese license plate with Darknet Yolo and converted its weight for Tensorflow.
I wrote a test program to confirm its detection performance in Python and built its docker image on my Windows PC.
The tensorflow frozen model data of Yolo V3 tiny is not quantized yet but its detection performance is 0.2 to 0.3 sec on my PC.
Now I want to confirm its performance on the P1467 cam.

As I have just started to develop the License Plate Recognition for AXIS cam, I am studying AXIS GitHub samples to run my Python test code. 
So I tried to build acap-native-sdk-examples/tensorflow-to-larod-artpec8 and object-detection and want to find the way to build .eap file for my testing on P1467 cam.
I uploaded the Docker Daemon on my P1467 cam.

Please be adviced for the next steps for my Yolo V3 tiny detection testing.
Thank you, 

Masataka

Cacsjep, 2024-12-21:
OK but you would only use Python right, to write your acap?

WilfR, 2024-12-21:
Hi @Ajiro3, 
  It sounds like you are on the right track. You will need a quantized version of your model in .tflite format. You can create that version of your model using whatever tools are familiar and convenient for you. 
  Once you have your model you will have to create an ACAP to use the model to analyze incoming video frames from the camera. This is not a trivial proposition, and not something I could answer here. Please make sure that you have read the documentation at https://developer.axis.com/computer-vision/. Probably the best course of action is to start with an existing example ACAP (like the object-detection example), get it running properly on your camera and then tailor it to use your model instead of the .tflite model that the example program is using. If your project is a learning experience this may be enough for your purposes. If you plan to build a real application you will have to add some feedback to the user: for example, show where in the video a license plate was detected, what the characters of the detected license plate are. For this you would have to use the Axis Overlay module in the Axis ACAP SDK. 
  I am not sure of your intention, but if you are installing Docker Daemon on the camera I am guessing that you are planning on creating a container-based ACAP.  If this is correct you may be interested to know that container-based applications are deprecated in the newest ACAP SDK, and "The recommended approach for developing analytics, computer vision, and machine learning applications on Axis devices is using the ACAP Native SDK.". See https://www.axis.com/developer-community/news/acap-12 for additional information. Again this may or may not be relevant to you depending on you plans for your ACAP application. 
  It sounds like a fun and interesting project to work on, Masataka. 
Good luck!
Cheers, 
Wilf.

Ajiro3, 2024-12-22:
Hi @Cacsjep,
yes, I use python not C nor ACAP.

Ajiro3, 2024-12-22:
Hi @WilfR,
Thank you for your kind advice.
What is the difference between [acap-computer-vision-sdk-examples](https://github.com/AxisCommunications/acap-computer-vision-sdk-examples) and https://github.com/AxisCommunications/acap-native-sdk-examples?
There seems to be some diffrence in their Dokerfile.
As I'm not familiar with docker, it is hard or me to understand their intension in Dockefile indeed.
Are there any good maerial to learn the how to write Dokerfile except ChatGPT?
Regards,
Masataka

Ajiro3, 2024-12-22:
Hi @WilfR,
I am stuked at the .pem files uploading to Cam.

I uploaded the pre-built docker daemon with AXIS IP Utility.

When I invoked it from its app list, an window was appeared and it said 
Upload TLS certificates and keys
curl --anyauth -u $DEVICE_USER:$DEVICE_PASSWORD -F file=@ca.pem -X POST http://$DEVICE_IP/local/dockerdwrapper/ca.pem

So, I tried to upload my created *.pem keys from Ubunts but got 404 response from Cam.
[image]

masataka@DESKTOP-I72467Q:~$ ls
ca-key.pem  ca.srl    get-pip.py  object-detection  server-key.pem
ca.pem      cert.csr  key.pem     server-cert.pem   server.csr

masataka@DESKTOP-I72467Q:~$ curl --anyauth -u root:Norikae4051 -F file=@ca.pem -X POST http://192.168.3.12/usr/local/packages/dockerdwrapper/localdata
<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>404 Not Found</title>
</head><body>
<h1>Not Found</h1>
<p>The requested URL was not found on this server.</p>
</body></html>
[image]

I chcked the target folder exists shown in the below.
root@axis-b8a44f61f646:/usr/local/packages/dockerdwrapper# ls
LICENSE                   dockerdwrapper            param.conf
cgi.conf                  html                      postinstallscript.sh
conf                      localdata                 ps
docker-init               manifest.json             rootlesskit
docker-proxy              nsenter                   rootlesskit-docker-proxy
dockerd                   package.conf              slirp4netns

Here is the log of DockerDaeon shown by AXIS IP Utility.
[image]

Will you kindly suggest me the way to solve this issue?
Thank you,
Masataka

WilfR, 2024-12-22:
Hi @Ajiro3, 
  "What is the difference between [acap-computer-vision-sdk-examples](https://github.com/AxisCommunications/acap-computer-vision-sdk-examples) and https://github.com/AxisCommunications/acap-native-sdk-examples?" 
  AXIS provides two different software development kits (SDKs) to build ACAP applications that run on Axis cameras. 
  1. The Computer Vision SDK. See https://github.com/AxisCommunications/acap-computer-vision-sdk for additional information. 
  2. The Native SDK. See https://github.com/AxisCommunications/acap-native-sdk for additional information. 
I have only used the Native SDK for ACAP development, and am not familiar with the Computer Vision SDK at all. Therefore I can't really say what are the differences between the acap-computer-vision-sdk-examples and the acap-native-sdk-examples repositories. 

"Are there any good maerial to learn the how to write Dokerfile except ChatGPT?" 
Docker is a complex system with many different use cases. For ACAP development the most important Docker command is "build" which uses a Dockerfile to create a Docker image. Docker "build" command is documented here : https://docs.docker.com/reference/cli/docker/buildx/build/. The format of the Dockerfile is documented here: https://docs.docker.com/reference/dockerfile/ .  I personally don't know of any Docker tutorials that are better than the tutorial offered by Docker itself : https://www.docker.com/101-tutorial/ . Again, for now you might wish to concentrate on parts of the tutorial that concentrate on building Docker image - publishing images, using volumes and other topics can wait for later. 

As far as the difficulties you've experienced with the .pem files I am afraid I am of no help now. I have never tried uploading a Docker daemon to camera, I have never used the Axis IP Utility and I don't even know what a .pem file is. So as I said I am afraid that I have no suggestions how to solve this current issue. 

Cheers, 
Wilf.

Cacsjep, 2024-12-22:
@Ajiro3 use computer vision sdk if you want use Python as language. Use native sdk when you want use C for development.
The documention and examples should be already be enough so it should be self explainable. Read the docs my friend ;)

Ajiro3, 2024-12-22:
Hi @WilfR,
Thank you for your help.
I found a way to upload the certification files(*.pem) with ACAP IP Utility.
Move to security > add certification, I can upload CA and other pem files.
Then I find them saved on /usr/local/packages/dockerdwrapper/localdata.
As dockerd is saved on /usr/local/packages/dockerdwrapper, but daemon.json in the folder is empty.
After I set parameters in daemon.json properly, I succeeded to invoke dockerd manualy.
But I can't invoke docker daemon from app list in ACAP IP Utility.
Regards,
Masataka

Ajiro3, 2024-12-22:
Hi @Cacsjep,
Thank you for your suggestion. 
Regards,
Masataka

closed 2024-12-22 by Ajiro3