AxisCommunications/acap-native-sdk-examples #319: Unable to load SSDLite MobileDet on ARTPEC-7

kossyrev-bg, 2024-12-30:
> **Please do not disclose security vulnerabilities as issues. See our [security policy](../../SECURITY.md) for responsible disclosures.**

### Describe the bug

I want to run the SSDLite MobileDet model for EdgeTPU, gotten from https://coral.ai/models/object-detection/, on my M3067-P axis cam. Instead, I'm getting the error shown in the logs below.

### To reproduce

Steps to reproduce the behavior:

1. Download the object detection example
2. Modify `manifest.json.edgetpu` with "runOptions": "/usr/local/packages/object_detection/model/converted_model.tflite 320 320 80 1920 1080 50 /usr/local/packages/object_detection/label/labels.txt -c axis-a7-gpu-tflite",
3. Download the SSDLite MobileDet model and labels from the website https://coral.ai/models/object-detection/
4. Modify the Dockerfile to copy those into the image instead of downloading each time (rename the files to `converted_model.tflite` and `labels.txt` on copy)
5. Modify the dockerfile to actually take in the `CHIP` arg (this seems like a separate bug you should address)
6. `docker build --build-arg ARCH=armv7hf --build-arg CHIP=edgetpu --tag obj_detect:1.0 . && docker cp $(docker create obj_detect:1.0):/opt/app/object_detection_edgetpu_1_0_0_armv7hf.eap ./build`
7. Go to camera site, apps, allow unsigned apps, upload and start the model
8. Note the error in the logs

### Environment

- Axis device model: M3067-P
- Axis OS: 11.11.73
- Logs from starting the app:
```
2024-12-30T14:29:59.041-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: Finding best resolution to use as model input
2024-12-30T14:29:59.180-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: chooseStreamResolution: We select stream w/h=320 x 320 based on VDO channel info.
2024-12-30T14:29:59.181-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: 'buffer.strategy': <uint32 3>
2024-12-30T14:29:59.181-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: 'channel'--------: <uint32 1>
2024-12-30T14:29:59.181-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: 'format'---------: <uint32 3>
2024-12-30T14:29:59.181-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: 'height'---------: <uint32 320>
2024-12-30T14:29:59.181-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: 'width'----------: <uint32 320>
2024-12-30T14:29:59.181-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: Creating VDO image provider and creating stream 320 x 320
2024-12-30T14:29:59.181-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: Dump of vdo stream settings map =====
2024-12-30T14:29:59.309-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: Find the best resolution to save the high resolution image
2024-12-30T14:29:59.389-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: 'buffer.strategy': <uint32 3>
2024-12-30T14:29:59.389-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: 'channel'--------: <uint32 1>
2024-12-30T14:29:59.389-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: 'format'---------: <uint32 3>
2024-12-30T14:29:59.389-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: 'height'---------: <uint32 1920>
2024-12-30T14:29:59.389-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: 'width'----------: <uint32 1920>
2024-12-30T14:29:59.389-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: chooseStreamResolution: We select stream w/h=1920 x 1920 based on VDO channel info.
2024-12-30T14:29:59.390-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: Creating VDO High resolution image provider and stream 1920 x 1920
2024-12-30T14:29:59.390-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: Dump of vdo stream settings map =====
2024-12-30T14:29:59.684-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: Calculate crop image
2024-12-30T14:29:59.684-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: Create larod models
2024-12-30T14:29:59.684-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: Create preprocessing maps
2024-12-30T14:29:59.684-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: Crop VDO image X=0 Y=0 (320 x 320)
2024-12-30T14:29:59.685-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: Setting up larod connection with chip axis-a7-gpu-tflite, model /usr/local/packages/object_detection/model/converted_model.tflite and label file /usr/local/packages/object_detection/label/labels.txt
2024-12-30T14:29:59.951-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: Available chip IDs:
2024-12-30T14:29:59.952-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: Chip: axis-a7-gpu-proc
2024-12-30T14:29:59.952-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: Chip: axis-a7-gpu-tflite
2024-12-30T14:29:59.952-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: Chip: axis-ace-proc
2024-12-30T14:29:59.952-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: Chip: cpu-proc
2024-12-30T14:29:59.952-05:00 axis-b8a44f4ba9eb [ INFO    ] object_detection[31464]: Chip: cpu-tflite
2024-12-30T14:30:00.138-05:00 axis-b8a44f4ba9eb [ ERR     ] object_detection[31464]: setupLarod: Unable to load model: Could not load model: Could not allocate tensors
2024-12-30T14:30:00.181-05:00 axis-b8a44f4ba9eb [ ERR     ] object_detection[31464]: Code: -2
```
- Version: Docker version 27.4.1, build b9d17ea (for building the image)

kossyrev-bg, 2024-12-30:
I guess maybe I'm confused. I assumed all ARTPEC-7 axis cameras had an edgeTPU, but then shouldn't that appear on the list of available chips? I couldn't find any info on the website https://www.axis.com/products/axis-m3067-p/support that stated whether it had a DLPU or not. All I saw was that it had ARTPEC-7, which from https://developer.axis.com/computer-vision/computer-vision-on-device/recommended-model-architecture I took to mean it had a DLPU built-in

Lallo, 2025-01-08:
Hi,
I believe you could use our product selector 
https://www.axis.com/support/tools/product-selector
looking for cameras with Artpec-7 SoC and DLPU compute capabilities (you will also find DLPU explicitly mentioned in their datasheet).

[image]

You could maybe use our Virtual Loan Tool to test a couple of them
https://www.axis.com/developer-community/axis-virtual-loan-tool

kossyrev-bg, 2025-01-09:
Thanks, I'll check it out.

closed 2025-01-09 by kossyrev-bg