AxisCommunications/acap-native-sdk-examples #320: Unable to load converted model to cpu-tflite

kossyrev-bg, 2025-01-09:
> **Please do not disclose security vulnerabilities as issues. See our [security policy](../../SECURITY.md) for responsible disclosures.**

### Describe the bug

Explain the behavior you would expect and the actual behavior.

### To reproduce

On an AXIS M3067-P, I attempted to load [RT-DETRv2](https://github.com/lyuwenyu/RT-DETR/tree/main/rtdetrv2_pytorch) converted to tflite, using the app build from `object_detection`. I immediately hit the `could not allocate tensors` error from calling the `larodLoadModel` function. The model is unfortunately too big to upload, but I will post a snapshot of netron.

Steps to reproduce the behavior:

1. Clone RT-DETRv2 repo
2. Download the v2 Small checkpoint
3. Install python requirements:
```
torch
torchvision
onnx
onnxruntime
tensorflow
tf_keras
pycocotools
PyYAML
scipy
psutil
onnx_graphsurgeon
IPython
onnx-simplifier
onnx2tf
simple_onnx_processing_tools
```
4. Run the below to export model to onnx:
```
python3 tools/export_onnx.py -c <path to repo>/rtdetrv2_pytorch/configs/rtdetrv2/rtdetrv2_r18vd_120e_coco.yml -r <path to downloaded model>/rtdetrv2_r18vd_120e_coco_rerun_48.1.pth --check
```
4.5. Note that I had to comment out the `dynamic_axes=dynamic_axes` step in the `export_onnx` file otherwise the next step would fail
5. Run onnx2tf to export the model to onnx:
```
onnx2tf -i <path to model>/model.onnx
```
6. This generates a folder of `saved_model`. 
7. Modify the Dockerfile in `object_detection` to import that model instead of downloading it from google. No need to modify labels file import; the model uses COCO labels
8. Build the app using `ARCH=armv7hf` and `CHIP=cpu`
9. Upload to the camera, start the app and see the failure

### Screenshots

[image]

### Environment

- Axis device model: M3067-P
- Axis device firmware version: 11-something

### Additional context

Add any other context about the problem here.

isak-jakobsson, 2025-02-13:
Hi @kossyrev-bg, could you attach the system log as well?

kossyrev-bg, 2025-02-13:
Sorry, this was long enough ago that I don't remember everything. I can see for example that the input tensors in the snippet i sent expected to be float32 instead of the uint8 that vdo outputs, so who knows what other problems were there. You can close this issue out.

closed 2025-02-14 by isak-jakobsson