AxisCommunications/acap-native-sdk-examples #321: Q: Possible to set environment variables when installing or running an application?

stefankiesz, 2025-01-17:
Hi, I have an application I'm running on the device that uses values from environment variables. I'm wondering if it is possible to set these for the application? I'm installing the application as a .eap file via the web interface.

I tried to export variables using `acapPackageConf.installation.postInstallScript` in the manifest, but it did not work, I'm guessing either because the script is not sourced or the application does not get run in the same shell.

Thanks!

pataxis, 2025-01-17:
Hi @stefankiesz , if it's a C/C++ application, have a look at the [axparameter](https://github.com/AxisCommunications/acap-native-sdk-examples/tree/main/axparameter) example for how you can work with parameters.

If it's a shell script application, I guess you could either set default values directly in the file or in a separate config file `myconfig.sh` 

```sh
export VAR1=yes
export VAR2=80
```

that is sourced by the main script. A separate file could be added to the .eap file with `acap-build . -a myconfig.sh`.

Reference: https://axiscommunications.github.io/acap-documentation/docs/develop/

stefankiesz, 2025-01-17:
Thanks @pataxis, I wasn't aware of the AXParameter library - very nice!