AxisCommunications/acap-native-sdk-examples #327: Using opensv example failing to install

AiVerisimilitude, 2025-01-22:
> **Please do not disclose security vulnerabilities as issues. See our [security policy](../../SECURITY.md) for responsible disclosures.**

### Describe the bug

Unable to install the "using-opencv" example. Through the web interface it will say "Unable to install" and through ssh it will say nothing. At least two other examples worked without issues (hello_world and vdostream) the rest were not attempted.

### To reproduce

No source etc was changed from the readme, but I will leave here the commands used for the ssh pipeline:

`docker build --progress=plain --no-cache --tag acap_usingopencv .`
`docker cp $(docker create acap_usingopencv):/opt/app ./build`
`scp build/opencv_example_1_0_0_armv7hf.eap root@<machine ip>:/tmp`

Then on the camera
`acap-install opencv_example_1_0_0_armv7hf.eap`

### Screenshots

Not necessary

### Environment

      "Architecture": "armv7hf",
      "ProdNbr": "Q6215-LE",
      "ProdFullName": "AXIS Q6215-LE PTZ Network Camera",
      "Version": "11.11.124",
      "ProdType": "PTZ Network Camera",
      "Soc": "Axis Artpec-6",
      "ProdShortName": "AXIS Q6215-LE",

SDK version set to the tag v1.15 according to the table in:
https://axiscommunications.github.io/acap-documentation/docs/axis-devices-and-compatibility/ 

The git log also mentions support of firmware version 11.11.70:
    Date:   Wed Jul 3 14:50:23 2024 +0200

    Update versions
    
    * Update VERSION to 1.15
    * Update UBUNTU_VERSION to 22.04
    * Update firmware version to 11.11.70
    * Update manifest schema version to 1.7.0

### Additional context

Nothing really. I'm just trying to get it to work with that example so that then I can build on top of it (mostly add gstreamer etc) but without any errors / sources I can take a look I am unable to proceed.

I could also not find a way to make the output of the install more verbose to get more information on why it's failing.

Cacsjep, 2025-01-22:
Please also send the camera log where it happens

AiVerisimilitude, 2025-01-23:
How can I get a log? There is no output to the console when using the `acap-install` command

Cacsjep, 2025-01-23:
Look Here, right to Report there the logs https://youtu.be/fU4yV0iyHCk

Cacsjep, 2025-01-23:
Please try to Install the app and the Copy Here the log

AiVerisimilitude, 2025-01-23:
Wohoo, some output at last :D thanks!

As for the log, I cannot put everything here yet since it might contain sensitive information but the offending lines seem to be these ones:

```
2025-01-22T11:03:54.793+00:00 axis-b8a44f1d0553 [ CRIT    ] acap-install[18469]: Invalid embedded development version "2.18". "opencv_app" require at least version 3.0.
2025-01-22T11:03:54.834+00:00 axis-b8a44f1d0553 [ WARNING ] packagemanager.cgi[17914]: pm_install_acap.c:98 pm_install_acap: Acap manager D-Bus install method returned error: tar: unexpected end of file tar: short read tar: unexpected end of file tar: short read tar: unexpected end of file tar: short read tar: unexpected end of file tar: short read (5)
2025-01-22T11:03:54.884+00:00 axis-b8a44f1d0553 [ WARNING ] packagemanager.cgi[17914]: pm_response.c:657 pm_response_send: Failed to write response to file descriptor 1: Broken pipe(32)
2025-01-22T11:03:54.924+00:00 axis-b8a44f1d0553 [ WARNING ] packagemanager.cgi[17914]: packagemanager.c:98 main: Failed to send error response
```

Seems to require the embedded dev version of 3.0? This is sound but makes the running the other 2 examples a bit puzzling since they also have that required version on their manifest.

How can I update the embedded dev version? is this part of the OS update or separate?

Cacsjep, 2025-01-23:
Hmm thats Sounds strange to me because it should be 3.x with that Firmware, that is a Thing axis devs must answer

Cacsjep, 2025-01-23:
Which example you use ? I cant find "using-opencv"

AiVerisimilitude, 2025-01-23:
I used this one:

https://github.com/AxisCommunications/acap-native-sdk-examples/tree/v1.15/using-opencv

AiVerisimilitude, 2025-01-23:
if its a Firmware issue I can try to do a rollback and re-update and see if something changes. Perhaps something went wrong during the update that was not visible. It will also give me a new set of logs ^_^

And thanks for the help so far!

Cacsjep, 2025-01-23:
Could you check If there something in the param list about Dev Version http://IP/axis-cgi/param.cgi?action=list

AiVerisimilitude, 2025-01-23:
That does not work on this one it seems. The log mentions some info but its just what we've found so far:
```
[Properties.EmbeddedDevelopment]
CacheSize                = "84934656"
DefaultCacheSize         = "100663296"
EmbeddedDevelopment      = "yes"
Version                  = "2.18"

...

[Properties.Firmware]
BuildDate                = "Nov 09 2024 17:00"
BuildNumber              = "1"
Version                  = "11.11.124"

[Properties.FirmwareManagement]
Version                  = "1.4"
```

What information are you trying to get from that command? perhaps there is another way to get to it

AiVerisimilitude, 2025-01-23:
Ah sorry, it was meant as a curl command, this is the output:

```
root.Audio.DSCP=0
root.Audio.DuplexMode=full
root.Audio.MaxListeners=20
root.Audio.MaxTransmitters=1
root.Audio.NbrOfConfigs=1
root.Audio.ReceiverBuffer=120
root.Audio.ReceiverTimeout=1000
root.Audio.A0.Enabled=no
root.Audio.A0.HTTPMessageType=singlepart
root.Audio.A0.Name=
root.Audio.A0.NbrOfChannels=1
root.Audio.A0.Source=0
root.AudioSource.NbrOfSources=0
root.AudioSource.A0.AlarmLevel=100
root.AudioSource.A0.AudioEncoding=aac
root.AudioSource.A0.BitRate=32000
root.AudioSource.A0.SampleRate=48000
root.AutoTracking.A0.Running=no
root.BasicDeviceInfo.BasicDeviceInfo=yes
root.Brand.Brand=AXIS
root.Brand.ProdFullName=AXIS Q6215-LE PTZ Network Camera
root.Brand.ProdNbr=Q6215-LE
root.Brand.ProdShortName=AXIS Q6215-LE
root.Brand.ProdType=PTZ Network Camera
root.Brand.ProdVariant=
root.Brand.WebURL=http://www.axis.com
root.HTTPS.AllowTLS1=no
root.HTTPS.AllowTLS11=no
root.HTTPS.Ciphers=ECDHE-ECDSA-AES128-GCM-SHA256:ECDHE-RSA-AES128-GCM-SHA256:ECDHE-ECDSA-AES256-GCM-SHA384:ECDHE-RSA-AES256-GCM-SHA384:ECDHE-ECDSA-CHACHA20-POLY1305:ECDHE-RSA-CHACHA20-POLY1305:DHE-RSA-AES128-GCM-SHA256:DHE-RSA-AES256-GCM-SHA384
root.HTTPS.Enabled=yes
root.HTTPS.Port=443
root.Image.NbrOfConfigs=1
root.Image.I0.Enabled=yes
root.Image.I0.Name=Video 1
root.Image.I0.Source=0
root.Image.I0.Appearance.Resolution=1920x1080
root.Image.I0.Appearance.Rotation=0
root.Image.I0.Stream.Duration=0
root.ImageSource.NbrOfSources=1
root.ImageSource.I0.CaptureFrequency=50Hz
root.ImageSource.I0.Name=Camera
root.ImageSource.I0.Sensor.AspectRatio=16:9
root.ImageSource.I0.Sensor.Stabilizer=off
root.ImageSource.I0.Video.DetectedType=HDTV1080p
root.Input.NbrOfInputs=0
root.Layout.AACInstallationEnabled=yes
root.Layout.DefaultJoystickMode=no
root.Layout.DefaultStreamProfile=
root.Layout.DefaultVideoFormat=mjpeg
root.Layout.EnableBasicSetup=yes
root.Layout.H264InstallationEnabled=yes
root.Layout.PlainConfigEnabled=no
root.Layout.ShowRelCrossEnabled=yes
root.Layout.ShowVideoFormatDropDown=yes
root.Layout.ViewerIE=activex
root.Layout.ViewerOther=spush
root.MediaClip.MaxGroups=10
root.MediaClip.MaxUploadSize=10
root.MediaClip.M9.Location=/etc/audioclips/camera_clicks16k.au
root.MediaClip.M9.Name=Camera clicks
root.MediaClip.M9.Type=audio
root.Network.Enabled=yes
root.Network.HTTP.AuthenticationPolicy=digest
root.Network.HTTP.AuthenticationWithQop=no
root.Network.IPv6.Enabled=no
root.Network.RTP.NbrOfRTPGroups=1
root.Network.RTSP.Port=554
[... 281 lines cut ...]
root.TemperatureControl.Heater.H7.ControlPolicyLimit=-99
root.TemperatureControl.Heater.H7.ManualControlDuration=3600
root.TemperatureControl.Heater.H7.UserControlSupport=yes
root.TemperatureControl.Sensor.S0.Name=S0
root.TemperatureControl.Sensor.S0.TriggerHigh=70
root.TemperatureControl.Sensor.S0.TriggerLow=-35
root.TemperatureControl.Sensor.S1.Name=S1
root.TemperatureControl.Sensor.S1.TriggerHigh=70
root.TemperatureControl.Sensor.S1.TriggerLow=-50
root.TemperatureControl.Sensor.S2.Name=S2
root.TemperatureControl.Sensor.S2.TriggerHigh=70
root.TemperatureControl.Sensor.S2.TriggerLow=-40
root.TemperatureControl.Sensor.S3.Name=S3
root.TemperatureControl.Sensor.S3.TriggerHigh=90
root.TemperatureControl.Sensor.S3.TriggerLow=-40
root.TemperatureControl.Sensor.S4.Name=S4
root.TemperatureControl.Sensor.S4.TriggerHigh=70
root.TemperatureControl.Sensor.S4.TriggerLow=-40
root.TemperatureControl.Sensor.S5.Name=S5
root.TemperatureControl.Sensor.S5.TriggerHigh=60
root.TemperatureControl.Sensor.S5.TriggerLow=-40
root.TemperatureControl.Sensor.S6.Name=S6
root.TemperatureControl.Sensor.S6.TriggerHigh=70
root.TemperatureControl.Sensor.S6.TriggerLow=-40
```

Cacsjep, 2025-01-23:
This one:  root.Properties.EmbeddedDevelopment.Version=2.18

Acap Want 3.0 but camera has 2.18

Could you try ubuntu Version 20.04 (docker)

AiVerisimilitude, 2025-01-23:
gonna try that

Cacsjep, 2025-01-23:
One Moment in manifest you have embeddedSdkVersion: "3" right

AiVerisimilitude, 2025-01-23:
Yep, that is the default version for the v1.15 of the native-sdk

Cacsjep, 2025-01-23:
The device should have a Higher Version Based on the compatilbe sheet

Cacsjep, 2025-01-23:
SDK 3.0	9.70 and later

Cacsjep, 2025-01-23:
Why yours is so low i dont know, because you have 11.x it should be higher as 3.5 regarding to the compatiblity tables

Cacsjep, 2025-01-23:
https://axiscommunications.github.io/acap-documentation/docs/acap-sdk-version-3/get-started/compatibility.html

AiVerisimilitude, 2025-01-23:
I did an upgrade to 11.x. Before it was 10.x version and according the upgrage plan I first upgraded to the latest 10.x version and the followed by an upgrate to the 11.x LTS since that was the last mentioned one for this device

Cacsjep, 2025-01-23:
I think the issue is Axis Artpec-6, thats a old SOC and looks not supported

Cacsjep, 2025-01-23:
Maybe it works when you Set the 2.18 in manifest

AiVerisimilitude, 2025-01-23:
> I think the issue is Axis Artpec-6, thats a old SOC and looks not supported

Nooooooo, that would be sad.

I will try setting the manifest to 2.18 with a new install. Though it will take a bit since the upload rate to the camera feels like its trying to upload it to the moon.

AiVerisimilitude, 2025-01-23:
it works! xD

Setting the embedded sdk version to 2.18 allows for the app to be installed. There seem to be no issues with running it as well as it produces believeble output

Many thanks!

Cacsjep, 2025-01-23:
Would be interessting If all the apis works, i think there must be a downside.

AiVerisimilitude, 2025-01-23:
Yeah, I hope not, but I wouldn't be surprised. Unfortunately, I doubt there is a way to test all APIs in one go, right? Eg. some sort of self-check stress test

It might be that future me will have to revisit this issue and see if there is something else that can be done. Though for now it works and that's a win :)

Cacsjep, 2025-01-23:
I dont know how old the cam is but there is a newer model of this camera AXIS Q6225-LE PTZ whats is Artpec 7 is there a reason why the old model?

Cacsjep, 2025-01-23:
> Yeah, I hope not, but I wouldn't be surprised. Unfortunately, I doubt there is a way to test all APIs in one go, right? Eg. some sort of self-check stress test
>
> It might be that future me will have [...]

You need to Test it by yourself but i dont have knowledge about that case where it is under 3.x

Cacsjep, 2025-01-23:
I think you are more in acap 3 with that camera with Like https://github.com/AxisCommunications/acap3-examples/tree/3.1

AiVerisimilitude, 2025-01-23:
> I dont know how old the cam is but there is a newer model of this camera AXIS Q6225-LE PTZ whats is Artpec 7 is there a reason why the old model?

It's what I have to work with xD. Perhaps if it becomes a blocker in the future, we will upgrade, but I will need to properly justify it as the process for getting new hardware is always a pain.

> I think you are more in acap 3 with that camera with Like https://github.com/AxisCommunications/acap3-examples/tree/3.1

Hmm, looks similar. Are there any major differences between the native sdk and acap3?

Cacsjep, 2025-01-23:
Some years of development ;) each change i dont know thats something you would need to research.

Cacsjep, 2025-01-23:
So think you can Close this ;)

AiVerisimilitude, 2025-01-23:
alright, thanks for everything!

closed 2025-01-23 by AiVerisimilitude

pataxis, 2025-01-24:
In manifest.json since AXIS OS 10.6, `embeddedSdkVersion` should be `3.0`, as the example and [documentation](https://axiscommunications.github.io/acap-documentation/docs/develop/application-project-structure.html#fn:2) show.

Is the manifest.json deleted by accident? Some of the output indicates that. You could check if it's in the .eap file by running
```sh
tar -tf opencv_example_1_0_0_armv7hf.eap
```

I recommend that you use the public VAPIX [upload.cgi](https://developer.axis.com/vapix/applications/application-api#upload-application) to install the application instead of using `scp` and `acap-install` which are not supported. With `curl`, depending on your network setup, something like this

```sh
curl -u<user>:<pass> --anyauth -F packfil=@opencv_example_1_0_0_armv7hf.eap <DEVICE_IP>/axis-cgi/applications/upload.cgi
```

You can also install by [running the build container](https://axiscommunications.github.io/acap-documentation/docs/develop/build-install-run.html#build-install-and-run-interactively-inside-container) interactively.

Cacsjep, 2025-01-24:
@pataxis the issue was that the Camera has embeeded Version 2.18

AiVerisimilitude, 2025-01-24:
Yes that's correct. I did an upgrade to 10.12.262 followed by an upgrade to 11.11.124, but that seems to not have updated the embedded SDK beyond 2.18