AxisCommunications/acap-native-sdk-examples #328: AXParameter "password" parameter type is not masked by "*"

stefankiesz, 2025-01-22:
### Describe the bug

When entering the value of a password parameter into the application settings in the Axis web interface, the characters are visible and are not masked by "*" as I would expect. The [AXParameter documentation](https://axiscommunications.github.io/acap-documentation/3.5/api/axparameter/html/ax__parameter_8h.html) states it should be:

>Rendered as an input field for string in the ACAP settings interface.
The value of the parameter is masked by '*'.

It _does_ change the saved characters to "*" after running or reinstalling the application, but when I type in a new value it is fully visible.

### To reproduce
I am testing on an Axis M3065-V (Ambarella S5L) device.

Steps to reproduce the behavior:

1. Modify the AXParameter example to have a password-type parameter.
2. Build and install onto the camera device.
3. Modify the password parameter in the web interface (tested on Chrome, Safari, and FireFox) and see that the characters are not masked.

### Screenshots
...

### Environment

- Axis device model: Axis M3065-V (Ambarella S5L)
- Axis device firmware version: 10.12
- OS and version: Mac OS 15.1.1
- Chrome, Safari, and FireFox browsers replicate the behavior.

pataxis, 2025-01-24:
Hi @stefankiesz , thanks for the report, this seems to be the same issue that was recently reported in https://github.com/orgs/AxisCommunications/discussions/875. A support case was opened and I think a fix is to be deployed in AXIS OS 12.3 as well as LTS service releases.

So check upcoming AXIS OS [release notes](https://help.axis.com/en-us/axis-os-release-notes). I will leave the ticket open for reference until the fix is confirmed.

pataxis, 2025-03-04:
Has been announced in 12.3 [release notes](https://help.axis.com/en-us/axis-os-release-notes#axis-os-12-3-56).

closed 2025-03-04 by pataxis