AxisCommunications/acap-native-sdk-examples #329: Type Mismatch Between Axis ONVIF XML Descriptions and Internal KVS Data

Cacsjep, 2025-01-24:
In my goxis library (see: [goxis repository](https://github.com/Cacsjep/goxis)), I encountered an issue where the ONVIF XML event definitions provided by Axis devices via ```get_event_list.py``` do not match the actual data types stored in the axevent kvs.

When retrieving values from the KVS using the expected types defined in the XML (e.g., tt:ReferenceToken), the system raises errors indicating a type mismatch. For example, fields expected to be **string** according to the xml ONVIF specification are stored as **int** in the KVS, causing unmarshalling errors.

```
2025-01-24T18:04:49.476+01:00 axis-b8a44f914a20 [ ERR     ] axeventmsubscribe[1758924]: Error unmarshalling DeviceTriggerDigitalInputEventKvs: error getting string for key InputToken: The value found is not of the expected type., AxEventError: AXEventErrorIncompatibleValue, Value Type: 0
2025-01-24T18:04:49.521+01:00 axis-b8a44f914a20 [ ERR     ] axeventmsubscribe[1758924]: Error unmarshalling DeviceHardwareFailurePowerSupplyFailurePTZPowerFailureEventKvs: error getting string for key Token: The value found is not of the expected type., AxEventError: AXEventErrorIncompatibleValue, Value Type: 0
2025-01-24T18:04:49.299+01:00 axis-b8a44f914a20 [ ERR     ] axeventmsubscribe[1758924]: Error unmarshalling DeviceTriggerRelayEventKvs: error getting string for key RelayToken: The value found is not of the expected type., AxEventError: AXEventErrorIncompatibleValue, Value Type: 0
2025-01-24T18:04:49.158+01:00 axis-b8a44f914a20 [ ERR     ] axeventmsubscribe[1758924]: Error unmarshalling VideoSourceMotionAlarmEventKvs: error getting string for key Source: The value found is not of the expected type., AxEventError: AXEventErrorIncompatibleValue, Value Type: 0
2025-01-24T18:04:49.189+01:00 axis-b8a44f914a20 [ ERR     ] axeventmsubscribe[1758924]: Error unmarshalling VideoSourceGlobalSceneChangeEventKvs: error getting string for key Source: The value found is not of the expected type., AxEventError: AXEventErrorIncompatibleValue, Value Type: 0
```

For example, the Source field in the MotionAlarm is defined as a tt:ReferenceToken, which should be a string:
```go
<MotionAlarm wstop:topic="true">
	<tt:MessageDescription IsProperty="true">
		<tt:Source>
			<tt:SimpleItemDescription Name="Source" Type="tt:ReferenceToken"></tt:SimpleItemDescription>
		</tt:Source>
		<tt:Data>
			<tt:SimpleItemDescription Name="State" Type="xsd:boolean"></tt:SimpleItemDescription>
		</tt:Data>
	</tt:MessageDescription>
</MotionAlarm>
```
However, querying this field from the KVS reveals that the stored type is int rather than the expected string. This results in the following error:

```
Error unmarshalling VideoSourceMotionAlarmEventKvs: error getting string for key Source: The value found is not of the expected type., AxEventError: AXEventErrorIncompatibleValue, Value Type: 0
```

AXIS M3215-LVE 12.0.91, embeddedSdkVersion: 3.5, manifest v 1.7.0

After switching the types to **int**, the errors are no longer present, confirming that the values are indeed stored as integers in all cases. While the XML definitions specify the type as **string**, the actual data retrieved from the KVS is of type **int**.

johan-olsson-work, 2025-05-08:
Hi.

We've looked into it and created an internal bug ticket so the responsible teams can work on a proper fix.

We'll update this issue when we have more information about the fix.

Thanks for bringing this to our attention and helping improve our platform!

johan-hultberg-work, 2025-08-20:
Hi. 

The internal bug ticket for this issue has been resolved and the fix has been merged into Axis OS, this means it till be available in the next Axis OS release, **Axis OS 12.6**.

We're keeping this issue open until we can verify that the fix resolves the problem completely. Once Axis OS 12.6 is released and we can confirm the solution works as expected, we'll close this issue.

Thank you again for the detailed bug report and for helping us improve the platform!

johan-hultberg-work, 2025-10-06:
I've confirmed that the fix in Axis OS 12.6 successfully resolves this issue - the error messages are no longer appearing when subscribing to events, and the type mismatch has been corrected.

Thank you for the detailed bug report and for helping us improve the platform.

Closing this issue as resolved in Axis OS 12.6.