AxisCommunications/acap-native-sdk-examples #332: axparameter - names with underscores are not editable via the interface

gustavofuhr, 2025-02-03:
### Describe the bug

When defining parameters in the `manifest.json` it appears that if you use underscore in the param name or the app name itself there is an error in the interface where every time you want to change some of the settings you got a "OPPs! Couldn't set the value"

### To reproduce
I just changed the manifest.json parameter of the axparameter sample from 
```
{
    "name": "IsCustomized",
    "default": "no",
    "type": "string"
},
```
```
{
    "name": "Is_Customized",
    "default": "no",
    "type": "string"
},
```
### Screenshots

[image]

### Environment

- Axis device model: AXIS P3268-LVE
- Axis device firmware version: 11.11.124

pataxis, 2025-02-04:
This should have been fixed in AXIS OS 12.0, it was mentioned in the ACAP 12.0 [release notes](https://axiscommunications.github.io/acap-documentation/docs/release-notes/12.0.html#bug-fixes).

pataxis, 2025-02-17:
@gustavofuhr  If I have correct information, the fix in AXIS OS 12.0 should be backported to AXIS OS LTS 11.11.
So keep an eye at the LTS 11.11 [release notes](https://help.axis.com/en-us/axis-os-release-notes#lts-2024-11-11).

Stiv-work, 2025-05-06:
Verified setting underscore in parameter in latest 11.11 LTS and it worked.

closed 2025-05-06 by Stiv-work