AxisCommunications/acap-native-sdk-examples #333: Attempting to add an existing parameter does not return the expected error code, but the error message is correct.

Cacsjep, 2025-02-05:
When I try to add a parameter that exists i got correclty ***The parameter is already added: <param_name>*** but the GError code is 13, according to [enum it should be 12](https://axiscommunications.github.io/acap-documentation/docs/acap-sdk-version-3/api/src/api/axparameter/html/ax__parameter__error_8h.html#ad8b86e23574970dce5ed65cca520cc37a06464953e09990a25403f3111dbc85e5)

thanks for checking

pataxis, 2025-02-12:
Hi, the link refers to ACAP 3 API doc, [here](https://developer.axis.com/acap/api/src/api/axparameter/html/ax__parameter__error_8h.html#ad8b86e23574970dce5ed65cca520cc37)'s the one for axparameter in ACAP Native SDK. The table seems unchanged in any case.

I'm counting and starting with 0, I get that the enum is 13 like you say the error report.

```txt
0. AX_PARAMETER_INVALID_ARG_ERROR
1. AX_PARAMETER_FILE_FD_ERROR
2. AX_PARAMETER_FILE_LOCK_ERROR 	
3. AX_PARAMETER_FILE_OPEN_ERROR 	
4. AX_PARAMETER_FILE_FORMAT_ERROR 	
5. AX_PARAMETER_FILE_CREATE_ERROR 	
6. AX_PARAMETER_FILE_WRITE_ERROR 	
7. AX_PARAMETER_FILE_LINK_ERROR 	
8. AX_PARAMETER_PARAM_LIST_ERROR 	
9. AX_PARAMETER_PARAM_GET_ERROR 	
10. AX_PARAMETER_PARAM_PATH_ERROR 	
11. AX_PARAMETER_PARAM_SYNC_ERROR 	
12. AX_PARAMETER_PARAM_EXIST_ERROR 	
13. AX_PARAMETER_PARAM_ADDED_ERROR 	Parameter was already added or had a preserved value 
```

Cacsjep, 2025-02-12:
ok but it should be 12 in case of param exists right ? =)

Cacsjep, 2025-02-12:
@pataxis oh boy, sorry for that, the name was just misleading, `AX_PARAMETER_PARAM_NOT_EXIST_ERROR` would be a better name 😅

thank you i close it =)

closed 2025-02-12 by Cacsjep