AxisCommunications/acap-native-sdk-examples #338: YOLOv5 Application Abnormalities After Several Days

quyetvkars, 2025-02-27:
We are developing an application using YOLOv5, but we are encountering abnormal behavior after the system has been running for several days. Below are the key issues:

Issue 1: License Key Verification Failure

- We have a function that verifies the license key daily by calling licensekey_verify().
- However, after several days, this function starts returning verification failed.
- The issue is temporarily resolved when we reset the camera, but it reoccurs after some time.

Issue 2: Failed Frame Fetching from VDO

- After a few days, we encounter the following error in our logs:

`threadEntry: Failed fetching frame from vdo: Socket Closed`

- The error occurs in the following code snippet:

```
static void* threadEntry(void* data) {
    GError* error = NULL;
    ImgProvider_t* provider = (ImgProvider_t*) data;

    while (!provider->shutDown) {
        // Block waiting for a frame from VDO
        VdoBuffer* newBuffer =
            vdo_stream_get_buffer(provider->vdoStream, &error);

        if (!newBuffer) {
            // Fail but we continue anyway hoping for the best.
            syslog(LOG_WARNING, "%s: Failed fetching frame from vdo: %s", __func__,
                       (error != NULL) ? error->message : "N/A");
            g_clear_error(&error);
            continue;
        }
    }
}
```

- We also observed the following message in the system logs:

`monolith[1162]: RTSP session uqBGHMSiKRP.G0A7 timed out.`

- In access log, RTSP session is created/terminted frequently. Is it normal?

```
monolith[1162]: RTSP video/x-h264 session V2UR6hO3zl5Uz7yB created from 127.1.1.2
monolith[1162]: RTSP video/x-h264 session V2UR6hO3zl5Uz7yB terminated from 127.1.1.2
monolith[1162]: RTSP video/x-h264 session yLLCubEhsGd0eurH created from 127.1.1.2
monolith[1162]: RTSP video/x-h264 session yLLCubEhsGd0eurH terminated from 127.1.1.2
```

**Investigation & Considerations**

We have ruled out the following possible causes:

- Memory leaks: No significant memory growth observed.
- Camera rotation: No unexpected movement detected.
- Network issues: Disconnected network does not impact to key verify
- Over limit parallel camera stream: there are only 3 camera streams at same time (stream to CCTV cloud, stream for AI operation, stream for web preview)

We are seeking insights into potential root causes and solutions for these issues. Any guidance or debugging suggestions would be greatly appreciated.

Cacsjep, 2025-02-27:
Which firmware and acap version ?

quyetvkars, 2025-02-27:
We are using firmware 12.1.64 and ACAP 3.5

Cacsjep, 2025-02-27:
So you use acap 12.1.0 ? (Docker) I had serval VDO acaps running and never got socket close issue. Does this also happen with the vdo example from repo ?

isak-jakobsson, 2025-03-13:
Hi!

Im sorry to hear that you are running into problems. Are there any specific reason why you are using ACAP 3.5?

You might be interested in the compatibility documentation of ACAP 3 available here https://developer.axis.com/acap/acap-sdk-version-3/get-started/compatibility/. To summarize, ACAP 3.5 is compatible with AXIS OS version 10.9 and later within the AXIS OS 10 LTS. This means that the APIs you are using might have introduced breaking changes both between AXIS OS 10 and 11, and also between AXIS OS 11 and 12. This is potentially causing the problems you're facing.

For building an ACAP application for AXIS OS 12.1.64, the supported SDK versions are 12.0.0 and 12.1.0.

isak-jakobsson, 2025-04-15:
Hi! Since we have not heard from you in a while, I'm going to close this issue. Feel free to reopen it if you need further assistance.

closed 2025-04-15 by isak-jakobsson