AxisCommunications/acap-native-sdk-examples #360: Per-tensor quantization

kossyrev-bg, 2025-03-20:
### Describe the bug

I have a tensorflow model that was produced from running onnx2tf on a torch model. I can load the model just fine, and I can even do per-channel quantization using TF2's `Interpreter` without a hitch. This converted model performs well.

Problem is, the M4328-P camera has an ARTPEC-8 chip which is much better with per-tensor quants. I attempted to use the `converter._experimental_disable_per_channel = True` flag, but then the outputted model just gave me the same value no matter what my input was. However, it does run about 10x faster on the camera, as expected given the per-tensor quantization layout. I'm trying to figure out how to debug. This is a mobilenetv1 variant, I can upload the model if necessary. Attached is a netron picture. 

### To reproduce

Load the model using `tf.lite.Interpreter(model)` and follow the standard steps to quantize it.

I have a slight suspicion that this per-tensor quantization will fail with any model.

### Screenshots

Per-channel quantized model top snippet:

[image]

Per-tensor quantized model snippet:

[image]

### Environment

- Axis device model: M4328-P
- Axis device firmware version: 11.11.135
- Stack trace or logs: [e.g. Axis device system logs]
- OS and version: Ubuntu 24.04, Python3.12, tensorflow 2.19
- Version: [version of the application, SDK, runtime environment, package manager, interpreter, compiler, depending on what seems relevant]

### Additional context

Add any other context about the problem here.

isak-jakobsson, 2025-03-21:
Hi @kossyrev-bg 
> Load the model using tf.lite.Interpreter(model) and follow the standard steps to quantize it.

Could you provide more details about how you did this? What standard steps do you refer to?

kossyrev-bg, 2025-03-21:
```
converter = tf.lite.TFLiteConverter.from_saved_model('saved_model')
converter.optimizations = [tf.lite.Optimize.DEFAULT]
converter.representative_dataset = representative_dataset
converter.target_spec.supported_ops = [tf.lite.OpsSet.TFLITE_BUILTINS_INT8]
converter.inference_input_type = tf.uint8
converter.inference_output_type = tf.float32
converter._experimental_disable_per_channel = True
tflite_quant_model = converter.convert()
```
I'm tempted to close out this ticket; I tried the exact same steps with the `mobilenetv2_100` variant and it seemed to work.

isak-jakobsson, 2025-04-03:
Im closing this issue. Feel free to reopen it if you still have problems.

closed 2025-04-03 by isak-jakobsson