AxisCommunications/acap-native-sdk-examples #361: How to declare an axevent with source choice like Virtual Input

Cacsjep, 2025-03-21:
Hey,

How could I create an axevent with choices like VirtualInput so users can select the source in an event rule ?
Do i need to use [ax_event_element ](https://axiscommunications.github.io/acap-documentation/docs/acap-sdk-version-3/api/src/api/axevent/html/ax__event__element__item_8h.html)? but there is the info "**_Using an AXEventElementItem as a data value enables an application to pass any data, that can be represented by a tree, as a value._**" so its only for data and not for an source selection ?

thanks

Endgoal would be when getting the event declarations that my event also have the "aev:Value" values.
```
<tns1:Device aev:NiceName="Device">
    <tnsaxis:IO aev:NiceName="Input ports">
	 <VirtualInput wstop:topic="true" aev:NiceName="Virtual input">
		 <aev:MessageInstance aev:isProperty="true">
			 <aev:SourceInstance>
				 <aev:SimpleItemInstance aev:NiceName="Port number" Type="xsd:int" Name="port">
					 <aev:Value>54</aev:Value>
					 <aev:Value>31</aev:Value>
					 <aev:Value>8</aev:Value>
					 <aev:Value>58</aev:Value>
					 <aev:Value>43</aev:Value>
```

isak-jakobsson, 2025-03-21:
Hi @Cacsjep
Sadly we don't have a clear answer to this question right now. We'll get back when we have had time to investigate it.

Cacsjep, 2025-03-21:
@isak-jakobsson 

Thx should I create a TIP case, would be answered faster?

BR

isak-jakobsson, 2025-04-03:
Hi @Cacsjep, I have some new information about this issue.

This can be achieved by declaring separate events with specific values for the Port field on each declaration. The other fields  should be kept the same.

Im still not sure what it means, but maybe it helps you.

Cacsjep, 2025-04-03:
ok thanks will try that and let you know if this works

closed 2025-04-06 by Cacsjep

Cacsjep, 2025-07-04:
Nice 🙂🙂🙂