AxisCommunications/acap-native-sdk-examples #368: Stateful event not always triggering MQTT actions

francesco-clementi-92, 2025-04-18:
Hi all and thanks for the examples.

I'm successfully using yolov5 on my camera to trigger axis event, which will send MQTT messages.

What I found weird is that in the camera logs I can see that I successfully sent the event, but sometimes the action rule to send the message it's not triggered.

In the camera station pro client I noticed that there is a flag called "event" that for some events seems to be necessary and for others not.

[image]

What does this flag do?

Is there a way in the code to trigger manually the action to send MQTT messages instead of relying on the events?

Cacsjep, 2025-04-21:
@francesco-clementi-92 

Looks like you had created a statefull event, that's why you have the option "Use this condition as trigger" basically you had a bool active entry in your event declaration and declare the event as statefull.

Also you have add a second bool flag called "event". 

The event declaration is up to the programmer and the use case how it's build up. 

In my opinion it made only sense for a statefull event to have one data entry, eg. active or inactive. 

In your scenario when you would send a message, when a train is detected you should either use a statefull event where you set the *active* when the train is detected and set *active* to false when train leave the scene. Or use stateless event where just the train detection work as an trigger.  

Use AXIS metadata monitor to verify the event sending, I found this quite handy for debugging.

Why it's not triggerd sometimes I can't answer because I don't know how you send the event in the code.

francesco-clementi-92, 2025-04-21:
Hi @Cacsjep,
correct, I set up a statefull event.

```
// setup configuration
ax_event_handler_declare(event_handler,
                                  key_value_set,
                                  FALSE,  // Stateful event
                                  &declaration,
                                  (AXDeclarationCompleteCallback)declaration_complete,
                                  &start_value,
                                  &error)
...
// event send
ax_event_handler_send_event(data->event_handler, 
                                    event_id, 
                                    event, 
                                    &error)
```

I agree that I should use only one statefull event, but it's not clear for me what that event flag means on the camera configuration page.

Cacsjep, 2025-04-21:
@francesco-clementi-92 please show the full kvs setup

Cacsjep, 2025-04-21:
@francesco-clementi-92 

We could also talk an discord when you like https://discord.gg/zF6mhmPajB

Cacsjep, 2025-05-06:
@francesco-clementi-92 I think we can close this or :)

francesco-clementi-92, 2025-05-07:
Apology, you are right! Thanks for the amazing support :)

closed 2025-05-07 by francesco-clementi-92